eModbus/eModbus #309: ModbusBridgeEthernet does not work

wind-bsk, 2023-07-06:
Hello! I am creating an application to convert ModBus TCP to Modbus RTU. The problem occurred when working with Ethernet. After the conversion task is started, it is completed. A message is sent to the serial port:
```
[1;33m[E] 5267| ModbusServerTCPtemp.h [ 252] serve: Server going down
[0m
```
Shutdown is not performed immediately, but after a certain period of time. Experimentally, it was found that the completion time is affected by a **delay** in file ModbusServerTCPtemp.h :
```
// start: create task with TCP server to accept requests
template <typename ST, typename CT>
  bool ModbusServerTCP<ST, CT>::start(uint16_t port, uint8_t maxClients, uint32_t timeout, int coreID) {
    // Task already running?
    if (serverTask != nullptr) {
      // Yes. stop it first
      stop();
    }
    // Does the required number of slots fit?
    if (numClients != maxClients) {
      // No. Drop array and allocate a new one
      delete[] clients;
      // Now allocate a new one
      numClients = maxClients;
      clients = new ClientData*[numClients]();
    }
    serverPort = port;
    serverTimeout = timeout;
    serverGoDown = false;

    // Create unique task name
    char taskName[18];
    snprintf(taskName, 18, "MBserve%04X", port);

    // Start task to handle the client
    xTaskCreatePinnedToCore((TaskFunction_t)&serve, taskName, 4096, this, 5, &serverTask, coreID >= 0 ? coreID : NULL);
    LOG_D("Server task %s started (%d).\n", taskName, (uint32_t)serverTask);

    // Wait two seconds for it to establish
    delay(2000);

    return true;
  }
```
If the delay is increased to 60,000 milliseconds, the conversion of Modbus TCP to ModBus RTU works correctly. After the delay expires, the conversion task is completed and the connection becomes unavailable.

Miq1, 2023-07-06:
While this is odd and requires more investigation, why don't you use a ``ModbusBridgeRTU``?

Some more details would be helpful to find the root cause. The line number 252 in your captured error message does not match that for the current code (245), so you either will have modified it or are using an older release. Second, this message is sent only if a ``stop()`` has been called before for the TCP server. You may have a close look at your sketch to find where that is done.

wind-bsk, 2023-07-06:
I can't understand why `stop()` is called in Modbus TCP Server. In line 233 reads the status of `myself->serverGoDown`. Someone assigns it the value `true `after executing `start()` and, accordingly, ModBusServerTCP is shut down.

````
Some more details would be helpful to find the root cause. The line number 252 in your captured error message does not match that for the current code (245) ...
````
I changed the delay value in line 166 and and while this delay is being performed, the task (Modbus Server TCP) is running.

wind-bsk, 2023-07-06:
I found a temporary solution. In the ModBusServerTCPtemp file.h in line 166 replaced: `delay(2000);` 
with
````
while(!serverGoDown){
      delay(1e5);
    }
````

Miq1, 2023-07-27:
Closing this as obviously is getting stale.

closed 2023-07-27 by Miq1