eModbus/eModbus #311: STM32 support

bobemoe, 2023-07-28:
This looks like a nice library, I'm thinking of trying to use it on STM32 Arduino.

Has anyone an idea of what would be involved? I guess its not as simple as just adding a new section to `options.h`?

Is there interest in a PR if I work on this?

Miq1, 2023-07-29:
You are welcome to adapt the lib to the STM32, but I am afraid it may not be that easy. With the exception of the async TCP stuff - that is running on ESP8266 et al. - the code makes heavy use of multitasking based on RTOS (or pthreads in the case of the Linux TCP client). This would have be migrated to whatever is possible on the STM32.

bobemoe, 2023-07-30:
Yeah I think you are right, I saw you could `#define HAS_FREERTOS 0` and hoped that enabled another implementation but now I've had a proper look at the code it appears to just disable the RTU functionality? 

I do believe STM32 has a FreeRTOS implementation but doesn't use it by default and I'm not using it in my project so I probably wont have time to work on this.

Thanks anyway. Feel free to close this or keep it open as you see fit.

Miq1, 2023-07-30:
``#define HAS_FREERTOS 0`` is swapping out everything except the AsyncTCP stuff (for ESP8266) and the Linux TCP client code.

closed 2023-07-30 by Miq1