eModbus/eModbus #318: Issue with ModBusTCP and data request

diegostf, 2023-10-20:
I have one ESP32 as Client and and three ESP32-S2 as Servers. 

Client make three requests in sequence (one for each device), to read 6 words (each device) and wait for the response from the 3 devices. All three devices respond in less than 100ms and the responses are OK.

But if Client requests 8 words from one device and 6 words from the others two devices, for example. The response from the two devices that were asked for 6 words is: error:0xEA, IP connection failed. The device that was requested 8 words responds normally.

If the client makes one request at a time (one request per device), even with a different number of words, everything goes well.

I hope I was clear.
Your library is fantastic.

Miq1, 2023-10-20:
That is odd. Modbus TCP, I suppose. The three servers will have different IP addresses, right? Then three independent TCP sessions are created, each being completely separated from the others in theory. So the length of TCP packets should not matter at all.

Can you provide us with a verbose log?

diegostf, 2023-10-20:
This is my request code:
```
void get_ModbusData(void){
   Error err;
   uint8_t index;  

   for(index=0; index<3; index++){
         tokenRequest++;
 
         IPAddress addressIp = ipMbTcp;          // Getclient IP
         addressIp[3] = mbAddDev[index];        // Change IP to servers. 10.0.1.11, 10.0.1.12 and 10.0.1.13. mbAddDev[] ={11, 12, 13}

         mbTCP.setTarget(addressIp, 502);   
          
         tokenKeyServer[index] = tokenRequest;  
         if(index == 2) err = mbTCP.addRequest(tokenKeyServer[index], 1, READ_HOLD_REGISTER, 1, 8);  
         else err = mbTCP.addRequest(tokenKeyServer[index], 1, READ_HOLD_REGISTER, 1, 6);    

         if(err != SUCCESS){
            ModbusError e(err);
            Serial.printf("[ MODBUS ] Error creating request: %02X - %s\n", (int)e, (const char *)e);
         }                                                   
    } 
}
```
 I'll provide a verbose log soon.

Miq1, 2023-10-25:
@diegostf Any news?

Miq1, 2023-11-01:
Looks like this is no longer relevant.

closed 2023-11-01 by Miq1