eModbus/eModbus #336: RTU Bus collisions with many devices on the same bus

raffaeler, 2024-02-14:
**Describe the bug**
I see a large number of CRC or Timeouts when more than 6-7 clients are connected to the bus.

* Library version: 1.7.0
* Hardware: ESP32 WROOM32
* Board: custom
* Transceivers: max33045e (supports up to 256 devices on the bus)
* Speed: 19200 (HardwareSerial)

**Expected behavior**
I have a bus with 1 client and 14 servers.
- When using just 6-7 servers (disconnecting the others), it works correctly
- With higher number of servers, there is an abundant and random number of CRC or Timeouts

**To Reproduce**
I reduced the code to just FC03 transmit/receive and I can see the issue only when a large number of devices are in the same bus.
To verify it only depends on the library, I removed the library and manually crafted modbus packets (both client and server) and I can send and receive without any issues.
I suspect the eModBus library is tricked by the answers of the other devices.

**Example code**
```
The reduced code is very close to the examples. The issue is visible only when a large number of devices lives on the same bus.
```

**Additional context**
I tried to use `skipLeading0x00` on both sides with worse results.
Unfortunately I don´t have any other clues to give.

raffaeler, 2024-02-14:
Additional note. With my own code, I raised the speed to 115K and works correctly.
Initially I did the test using 11K with eModbus and I got tons of errors.
Therefore, this could be a simple way to reproduce the issue without adding a large number of devices. Anyway, the transceiver should be adequate to the desired speed.

Miq1, 2024-02-14:
Speed is no issue I reckon. I have run RTU up to the ESP32's UART limit of approx. 5 Mbaud with no issues.
I would like to see the verbose log to learn what is going on on your bus.

raffaeler, 2024-02-14:
It requires some time to put back the library at work. BTW when I took a look at the log, I could see mixed garbage, it was not of great help. I also used QModBus apps (master, slave, monitor) and it was totally useless.

How many devices can you put together in your "lab"? This was the key to start seeing errors. This would be far easier to understand the issue because you can add/remove devices until you see the garbage.

My guess is that there is an issue on the timing between RTU frames. This way, the buffer with data coming from other servers is not discarded and bytes get mixed.

Miq1, 2024-02-14:
My assumptions are going the same way, but I rather would think of two possible external causes: servers not maintaining the minimum quiet time or RE/DE switching issues injecting stray 0x00 bytes. 
The maximum number of real servers I can provide ad hoc is 4 - no more hardware in stock I am afraid. I can simulate more servers with a single MCU, but that of course is a different thing.

raffaeler, 2024-02-14:
Given the time needed to update all devices, I'll have just a single shot to produce the logs from the client and a single server.
Please tell me the exact options for dumping the verbose log you want, because immediately after posting the log I will have to change strategy for the project and will have no more time to make further tests. (Sorry, but It is taking too long)

Miq1, 2024-02-14:
The client log will be the most relevant (see further down for an addition), so we need to have it compiled with ``-DLOG_LEVEL=6``. If you are using the Arduino IDE the easiest way to accomplish that is to modify the default ``LOG_LEVEL`` in the ``Logging.h`` file.

If you have some more resources, you can set up another server as a [sniffing device](https://emodbus.github.io/modbusserver-rtu-api#void-registersniffermsrlistener-worker) . You will get every single message on the bus and may print it out or do whatever else you like with it. This may help to see if the behaviour is identical to the client in terms of error messages etc.

Miq1, 2024-02-15:
@raffaeler Any results yet?

raffaeler, 2024-02-15:
I am restoring back eModbus now to create the logs.
After this, I'll have to go on with my code because I exhausted the time while looking for the problem :(
I'll post the logs in a few minutes

raffaeler, 2024-02-15:
The log is too verbose and I don´t understand if it is doing wrong or not

The client:
```
[D] 6163| ModbusClientRTU.cpp  [ 112] setTimeout: Timeout set to 2000
[V] 6164| RTUutils.cpp         [ 112] calculateInterval: Calc interval(115201)=1750
[D] 6174| ModbusClientRTU.cpp  [  86] doBegin: Client task 1073554272 started. Interval=1750
ReadySending the configuration: NewProdF2
[V] 15004| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15004| ModbusClientRTU.cpp  [ 156] addRequestM: request for 69/10
[D] 15015| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15015| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[D] 15016| ModbusClientRTU.cpp  [ 257] handleConnection: Pulled request from queue
[V] 15025| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] Sent packet: @3FFCED70/9:
  | 0000: 69 10 00 32 00 01 02 00  01                       |i..2.....       |
[D] 15036| ModbusClientRTU.cpp  [ 156] addRequestM: request for 6E/10
[D] 15046| ModbusClientRTU.cpp  [ 262] handleConnection: Request sent.
[D] 15056| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15066| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15066| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15077| ModbusClientRTU.cpp  [ 156] addRequestM: request for 6F/10
[D] 15077| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15087| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15087| ModbusMessage.cpp    [ 519] checkData: Check data #5
[V] 15094| RTUutils.cpp         [ 262] receive: C/1754us without data after 6
[V] 15098| RTUutils.cpp         [ 272] receive: C/[D] 15098| ModbusClientRTU.cpp  [ 156] addRequestM: request for 6B/10
[V] Raw buffer received: @3FFD2664/6:
[D] 15108| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
  | 0000: 69 10 00 01 DD BD                                 |i.....          |
[D] 15118| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15129| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15129| RTUutils.cpp         [ 437] receive: C/[D] Received packet: @3FFCF004/4:
[D] 15139| ModbusClientRTU.cpp  [ 156] addRequestM: request for 6C/10
  | 0000: 69 10 00 01                                       |i...            |
[D] 15149| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15160| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[D] 15160| ModbusClientRTU.cpp  [ 277] handleConnection: Data response (4 bytes) received.
[V] 15170| ModbusMessage.cpp    [ 519] checkData: Check data #5
[V] Data: @3FFCF004/4:
  | 0000: 69 10 00 01                                       |i...            |
[D] 15180| ModbusClientRTU.cpp  [ 156] addRequestM: request for 70/10
[D] 15191| ModbusClientRTU.cpp  [ 298] handleConnection: Response generated.
[V] Response packet: @3FFCF004/4:
  | 0000: 69 10 00 01                                       |i...            |
[D] 15201| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15211| ModbusClientRTU.cpp  [ 257] handleConnection: Pulled request from queue
[D] 15221| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[D] Sent packet: @3FFD05C0/9:
[V] 15232| ModbusMessage.cpp    [ 519] checkData: Check data #5
  | 0000: 6E 10 00 32 00 01 02 00  02                       |n..2.....       |
[D] 15242| ModbusClientRTU.cpp  [ 262] handleConnection: Request sent.
[D] 15242| ModbusClientRTU.cpp  [ 156] addRequestM: request for 71/10
[D] 15253| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15263| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15263| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15273| ModbusClientRTU.cpp  [ 156] addRequestM: request for 72/10
[D] 15274| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15284| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15284| ModbusMessage.cpp    [ 519] checkData: Check data #5
[V] 15290| RTUutils.cpp         [ 262] receive: C/1756us without data after 6
[D] 15294| ModbusClientRTU.cpp  [ 156] addRequestM: request for 65/10
[V] 15304| RTUutils.cpp         [ 272] receive: C/[D] 15305| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[V] Raw buffer received: @3FFD2664/6:
[D] 15315| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
  | 0000: 6E 10 00 01 DC C9                                 |n.....          |
[V] 15325| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15336| RTUutils.cpp         [ 437] receive: C/[D] 15336| ModbusClientRTU.cpp  [ 156] addRequestM: request for 66/10
[D] Received packet: @3FFCF248/4:
  | 0000: 6E 10 00 01                                       |n...            |
[... 409 lines cut ...]
  | 0000: 6D 10 00 01                                       |m...            |
[D] 835872| ModbusServerRTU.cpp  [ 173] serve: Request received.
[D] 835872| ModbusServer.cpp     [  44] getWorker: No matching worker found
[V] 835917| RTUutils.cpp         [ 262] receive: S/1752us without data after 11
[V] 835917| RTUutils.cpp         [ 272] receive: S/[V] Raw buffer received: @3FFD3EB8/11:
  | 0000: 6A 10 00 32 00 01 02 00  02 F8 B1                 |j..2.......     |
[D] 835928| RTUutils.cpp         [ 437] receive: S/[D] Received packet: @3FFB98A0/9:
  | 0000: 6A 10 00 32 00 01 02 00  02                       |j..2.....       |
[D] 835948| ModbusServerRTU.cpp  [ 173] serve: Request received.
[D] 835949| ModbusServer.cpp     [  44] getWorker: No matching worker found
[V] 835982| RTUutils.cpp         [ 262] receive: S/1773us without data after 6
[V] 835982| RTUutils.cpp         [ 272] receive: S/[V] Raw buffer received: @3FFD3EB8/6:
[... cut ...]
[D] 836013| ModbusServer.cpp     [  44] getWorker: No matching worker found
[D] 838025| RTUutils.cpp         [ 437] receive: S/[D] Received packet: @3FFB9874/1:
  | 0000: E0                                                |.               |
[D] 840037| RTUutils.cpp         [ 437] receive: S/[D] Received packet: @3FFB9798/1:
  | 0000: E0                                                |.               |
[D] 842049| RTUutils.cpp         [ 437] receive: S/[D] Received packet: @3FFB9910/1:
  | 0000: E0                                                |.               |

```

raffaeler, 2024-02-15:
These other logs are generated with everything at 19.2K baud.
The hardware configuration is always the same (14 servers, 1 client) as described in a previous message.

client: 
```
[D] 6164| ModbusClientRTU.cpp  [ 112] setTimeout: Timeout set to 2000
[V] 6164| RTUutils.cpp         [ 112] calculateInterval: Calc interval(19200)=1822
[D] 6175| ModbusClientRTU.cpp  [  86] doBegin: Client task 1073554296 started. Interval=1822
ReadySending the configuration: NewProdF2
[V] 15004| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15004| ModbusClientRTU.cpp  [ 156] addRequestM: request for 69/10
[D] 15015| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15015| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[D] 15016| ModbusClientRTU.cpp  [ 257] handleConnection: Pulled request from queue
[V] 15025| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] Sent packet: @3FFD037C/9:
  | 0000: 69 10 00 32 00 01 02 00  01                       |i..2.....       |
[D] 15036| ModbusClientRTU.cpp  [ 156] addRequestM: request for 6E/10
[D] 15046| ModbusClientRTU.cpp  [ 262] handleConnection: Request sent.
[D] 15056| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15066| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15066| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15077| ModbusClientRTU.cpp  [ 156] addRequestM: request for 6F/10
[D] 15077| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15087| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15087| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15098| ModbusClientRTU.cpp  [ 156] addRequestM: request for 6B/10
[D] 15098| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15108| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15108| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15119| ModbusClientRTU.cpp  [ 156] addRequestM: request for 6C/10
[D] 15119| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15129| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15129| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15140| ModbusClientRTU.cpp  [ 156] addRequestM: request for 70/10
[D] 15140| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15150| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15150| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15161| ModbusClientRTU.cpp  [ 156] addRequestM: request for 71/10
[D] 15161| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15171| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15171| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15182| ModbusClientRTU.cpp  [ 156] addRequestM: request for 72/10
[D] 15182| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15192| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15192| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15203| ModbusClientRTU.cpp  [ 156] addRequestM: request for 65/10
[D] 15213| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15213| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15224| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15224| ModbusClientRTU.cpp  [ 156] addRequestM: request for 66/10
[D] 15234| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15234| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15245| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15245| ModbusClientRTU.cpp  [ 156] addRequestM: request for 67/10
[D] 15255| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15255| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15266| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15266| ModbusClientRTU.cpp  [ 156] addRequestM: request for 68/10
[D] 15276| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15276| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15287| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15287| ModbusClientRTU.cpp  [ 156] addRequestM: request for 6D/10
[D] 15297| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15297| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15308| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15308| ModbusClientRTU.cpp  [ 156] addRequestM: request for 6A/10
[D] 15318| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15318| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[D] 17057| RTUutils.cpp         [ 437] receive: C/[D] Received packet: @3FFCF01C/1:
[... 144 lines cut ...]
  | 0000: 00 E0 00 00 00 00 00 00  00 E0 00 00 00 00 E0 00  |................|
  | 0010: E0 E0 00 00 E0                                    |.....           |
[D] 1116926| RTUutils.cpp         [ 437] receive: S/[D] Received packet: @3FFB9798/1:
  | 0000: E2                                                |.               |
[1B][1;33m[E] 1116946| ModbusServerRTU.cpp  [ 252] serve: RTU receive: E2 - CRC check error
[1B][0m[D] 1118948| RTUutils.cpp         [ 437] receive: S/[D] Received packet: @3FFB980C/1:
  | 0000: E0                                                |.               |
[V] 1118975| RTUutils.cpp         [ 262] receive: S/1753us without data after 23
[V] 1118975| RTUutils.cpp         [ 272] receive: S/[V] Raw buffer received: @3FFD43BC/23:
  | 0000: E0 00 E0 00 00 00 00 00  00 00 E0 00 00 00 00 00  |................|
  | 0010: 00 00 00 E0 E0 00 E0                              |.......         |
[D] 1118996| RTUutils.cpp         [ 437] receive: S/[D] Received packet: @3FFB9798/1:
  | 0000: E2                                                |.               |
[1B][1;33m[E] 1119017| ModbusServerRTU.cpp  [ 252] serve: RTU receive: E2 - CRC check error
[... cut ...]
[V] 1121046| RTUutils.cpp         [ 272] receive: S/[V] Raw buffer received: @3FFD43BC/24:
  | 0000: 00 00 E0 00 00 00 00 00  00 00 E0 00 00 00 00 00  |................|
  | 0010: 00 00 00 E0 E0 E0 00 E0                           |........        |
[D] 1121067| RTUutils.cpp         [ 437] receive: S/[D] Received packet: @3FFB9798/1:
  | 0000: E2                                                |.               |
[1B][1;33m[E] 1121088| ModbusServerRTU.cpp  [ 252] serve: RTU receive: E2 - CRC check error
[1B][0m
```

raffaeler, 2024-02-15:
@Miq1 Could you please confirm this is enough for you?

Miq1, 2024-02-15:
The server log: yes! The client log is not a verbose one, for whatever reason...

raffaeler, 2024-02-15:
I reflashed the client and here we are again with the 19.2K logs

client:
```
[V] 15004| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15004| ModbusClientRTU.cpp  [ 156] addRequestM: request for 69/10
[D] 15015| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15015| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[D] 15016| ModbusClientRTU.cpp  [ 257] handleConnection: Pulled request from queue
[V] 15025| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] Sent packet: @3FFCF158/9:
  | 0000: 69 10 00 32 00 01 02 00  01                       |i..2.....       |
[D] 15036| ModbusClientRTU.cpp  [ 156] addRequestM: request for 6E/10
[D] 15046| ModbusClientRTU.cpp  [ 262] handleConnection: Request sent.
[D] 15046| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15066| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15067| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15077| ModbusClientRTU.cpp  [ 156] addRequestM: request for 6F/10
[D] 15077| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15087| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15088| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15098| ModbusClientRTU.cpp  [ 156] addRequestM: request for 6B/10
[D] 15098| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15108| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15109| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15119| ModbusClientRTU.cpp  [ 156] addRequestM: request for 6C/10
[D] 15119| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15129| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15130| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15140| ModbusClientRTU.cpp  [ 156] addRequestM: request for 70/10
[D] 15140| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15150| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15151| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15161| ModbusClientRTU.cpp  [ 156] addRequestM: request for 71/10
[D] 15161| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15171| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15172| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15182| ModbusClientRTU.cpp  [ 156] addRequestM: request for 72/10
[D] 15182| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15192| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15193| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15203| ModbusClientRTU.cpp  [ 156] addRequestM: request for 65/10
[D] 15213| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15213| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15214| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15224| ModbusClientRTU.cpp  [ 156] addRequestM: request for 66/10
[D] 15234| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15234| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15245| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15245| ModbusClientRTU.cpp  [ 156] addRequestM: request for 67/10
[D] 15255| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15255| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15266| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15266| ModbusClientRTU.cpp  [ 156] addRequestM: request for 68/10
[D] 15276| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15277| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15287| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15287| ModbusClientRTU.cpp  [ 156] addRequestM: request for 6D/10
[D] 15297| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15298| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[V] 15308| ModbusMessage.cpp    [ 519] checkData: Check data #5
[D] 15308| ModbusClientRTU.cpp  [ 156] addRequestM: request for 6A/10
[D] 15318| ModbusClientRTU.cpp  [ 239] addToQueue: RC=01
[D] 15319| ModbusClientRTU.cpp  [ 167] addRequestM: RC=00
[D] 17057| RTUutils.cpp         [ 437] receive: C/[D] Received packet: @3FFCF104/1:
  | 0000: E0                                                |.               |
[D] 17068| ModbusClientRTU.cpp  [ 277] handleConnection: Error response (1 bytes) received.
[V] Data: @3FFCF104/1:
  | 0000: E0                                                |.               |
[D] 17079| ModbusClientRTU.cpp  [ 298] handleConnection: Response generated.
[... 141 lines cut ...]
  | 0000: 71 10 00 32 00 01 02 00  02 46 41                 |q..2.....FA     |
[D] 22714| RTUutils.cpp         [ 437] receive: S/[D] Received packet: @3FFD3A94/9:
  | 0000: 71 10 00 32 00 01 02 00  02                       |q..2.....       |
[D] 22734| ModbusServerRTU.cpp  [ 173] serve: Request received.
[D] 22734| ModbusServer.cpp     [  44] getWorker: No matching worker found
[D] 24746| RTUutils.cpp         [ 437] receive: S/[D] Received packet: @3FFD3AB4/1:
  | 0000: E0                                                |.               |
[V] 24774| RTUutils.cpp         [ 262] receive: S/1829us without data after 11
[V] 24774| RTUutils.cpp         [ 272] receive: S/[V] Raw buffer received: @3FFD3FD4/11:
  | 0000: 72 10 00 32 00 01 02 00  02 52 B1                 |r..2.....R.     |
[D] 24784| RTUutils.cpp         [ 437] receive: S/[D] Received packet: @3FFD3A94/9:
  | 0000: 72 10 00 32 00 01 02 00  02                       |r..2.....       |
[D] 24805| ModbusServerRTU.cpp  [ 173] serve: Request received.
[D] 24805| ModbusServer.cpp     [  44] getWorker: No matching worker found
[D] 26817| RTUutils.cpp         [ 437] receive: S/[D] Received packet: @3FFD3AB4/1:
[... cut ...]
[V] 26844| RTUutils.cpp         [ 262] receive: S/1829us without data after 11
[V] 26844| RTUutils.cpp         [ 272] receive: S/[V] Raw buffer received: @3FFD3FD4/11:
  | 0000: 65 10 00 32 00 01 02 00  01 F9 40                 |e..2......@     |
[D] 26854| RTUutils.cpp         [ 437] receive: S/[D] Received packet: @3FFD3A94/9:
  | 0000: 65 10 00 32 00 01 02 00  01                       |e..2.....       |
[D] 26875| ModbusServerRTU.cpp  [ 173] serve: Request received.
[D] 26875| ModbusServer.cpp     [  44] getWorker: No matching worker found

```

Miq1, 2024-02-15:
Looks pretty okay to me with one exception - see down below.

The client is firing requests to a couple of servers, but with the exception of server 0x6E none is responding at all. So obviously those requests will result in a 0xE0 TIMEOUT error response. 

The server log shows all the requests the client is sending, but of course only the requests addressed to its own server ID are processed and the response is sent.

The odd thing are the 0xE0 bytes on the bus the server log is showing. Since there is no "Raw buffer" log line before these, the code must have been generated internally for irrelevant requests. No harm done in fact, but I will see if I can omit these from the logic.

Remains the question why the other servers are playing dead?

Miq1, 2024-02-15:
I eventually will replace the internal 0xE0 error code with another to signal a "no data seen in the meantime" state to avoid confusion in the logs. But as I wrote: purely internal with no consequences at all.

Miq1, 2024-02-15:
Sorry I could not resolve your problem 😟 

As I wrote the logs are consistent and are showing no unexpected behaviour. There was no CRC error in these logs, only TIMEOUTs on the client log that are in line with the server's log due to no other server was responding.

raffaeler, 2024-02-15:
No worries, thank you anyway!

closed 2024-02-15 by raffaeler

I-Connect, 2024-02-15:
Just raised this issue: https://github.com/eModbus/eModbus/issues/337

Id did not read this issue on all detail but the more devices you have the more chance the same happens as what I am experiencing, you could try adding the delay...

raffaeler, 2024-02-15:
> Id did not read this issue on all detail but the more devices you have the more chance the same happens as what I am experiencing, you could try adding the delay...

Delay is not an option if you want to stay inside the standard, otherwise you could have problem attaching 3rd party devices to your bus.
The modbus protocol prescribes strict distances among packets (3.5 ... 4.5 characters).
This is needed because modbus packets do not contain a length indicator as it happens in many other protocols. Some function codes do not allow to "predict" the length of the packet. In other words the servers must recognize packets because they are well "spaced".

I currently drive my own client and servers (I have 14 servers and 1 client which will grow) and there are no issues at all at 115K. I will search the upper speed limit soon, but the speed depends on the cpu speed used to sample/poll the line, the transceiver quality and the cable.

raffaeler, 2024-02-15:
@Miq1 For your info (it is just a suggestion, not a criticism), I noticed that the code using eModBus requires more current than my (simpler) current implementation. I noticed it because I measure the temperature on the linear regulator.
Next time you put your hands on the code, you may want to consider this aspect as well.
HTH

Miq1, 2024-02-15:
The bug @I-Connect found will affect the next request a client is sending after a successfully received response. This request may be sent without maintaining the required quiet time, if the client's onData handler will take less than that to finish. We are talking about less than two milliseconds here (for baud rates 19k2 and above), so it will not happen all the time.

Miq1, 2024-02-15:
> @Miq1 For your info (it is just a suggestion, not a criticism), I noticed that the code using eModBus requires more current than my (simpler) current implementation. I noticed it because I measure [...]

I see few options for that, if not sacrificing responsiveness for less power consumption. The Arduino way would be to attach an interrupt to the RX line and sleep until a byte will arrive, but the ESP32 does that internally in the core already.

raffaeler, 2024-02-15:
> The Arduino way would be to attach an interrupt to the RX line and sleep until a byte will arrive, but the ESP32 does that internally in the core already.

In theory yes, but don't do it.
Unfortunately there is a known issue on the ESP32 HardwareSerial wrapper when using the default pins. In this case the event is not raised. But if you change the pins, it works.

On my side, I started a low-priority task polling the line and calibrating the timeouts so that if you try to read beyond the Modbus "gap", you get a timeout and know that the packet is finished. I also added a few safe delays in terms of hundreds of microseconds (proportional to the baud rate of course) when you know that nothing useful can be on the line.
Doing that I never got a single error on the line on both 19.2K and 115K.

Miq1, 2024-02-15:
Thanks, I will consider that. 👍🏻