eModbus/eModbus #337: Timeouts when polling 2 different Modbus devices right after one another

I-Connect, 2024-02-15:
**Describe the bug**
When polling 2 different Modbus devices directly after one another without any delay the second requests times out. (in synchronous mode)

So I put two read requests in the queue without any delay and 1 of them times out

[image]

**Expected behavior**
No timeout should happen and a value should be returned

**To Reproduce**
Put two read requests for 2 different devices in the queue without any delay (it has to be 2 different devices, it never occurred for me with high speed concurrent polling of just 1 device)
Disable any logging that could interfere with timing of pulling requests from the queue

**Additional context**
When searching for Modbus message timing I find several articles related to this:

[image]

or 
https://github.com/ioBroker/ioBroker.modbus/issues/39

It seems there needs to be some time between receiving a response and sending the next request

void ModbusClientRTU::handleConnection(ModbusClientRTU* instance)
Does contain a delay but that is only executed when there is no request in the queue.

I was able to resolve this by adding a 10ms (5ms is too little) delay between pulling requests from the queue in ModbusClientRTU.cpp line 345
[image]

[image]

(Not sure but this could also be an issue within Modbus TCP, although this works a bit different)

Miq1, 2024-02-15:
This is odd... It definitely can happen on RTU only, as the "quiet time" is only specified for Modbus RTU. 
I would have sworn all oaths I had the necessary microseconds delay implemented. I will have a look at it tomorrow.

Miq1, 2024-02-15:
I could not wait and had a look. OMG, you are right, the quiet time code has ominously disappeared without me noticing it! 😱 
 Will fix that soon in the "master" branch and even may have a release for it.

I-Connect, 2024-02-16:
👍

Miq1, 2024-02-16:
This seems to be more complicated than I first thought (and is an explanation for the seemingly missing quiet time delay):
1. We send a request.
2. We will wait for response data. Two alternatives:
    1. There is data coming. Now the receive loop will wait for the interval time after the last byte received to make sure the transmission is complete. ==> no quiet time violation here.
    2. There is no data at all. The loop now will terminate when the set timeout has passed - much longer than the quiet time interval would require ==> no quiet time violation here as well!

So what could be happening instead?

Miq1, 2024-02-16:
@I-Connect : can we have a log from the eModbus lib please? You seem to have rolled your own, but the internal log will give more detail and, more important, milliseconds timing.

I-Connect, 2024-02-16:
I removed the delay I added earlier and ran with your logging enabled (-DLOG_LEVEL=LOG_LEVEL_VERBOSE) but this probably changes timing so much that the issue does not occur any more :-(

[emodbus log.txt](https://github.com/eModbus/eModbus/files/14310895/emodbus.log.txt)

Info: I am reading in total 8 registers from 2 devices (kWh meters) at 9600 baud. 4 of these registers are read every second

I removed the log verbose define and reduced the logging to these just two lines:

[image]

And you see the errors popping up (which in the end leads to a queue full as the timeout takes 2 secs and I have a higher polling frequency)

So it takes about  29 ms for the read and (within) 1 ms after the read is finished the next request is pulled from the queue. No quiet time...?

Let me know if you want me to log something specific

```
[ 19964][D][ModbusClientRTU.cpp:259] handleConnection(): Pulled request from queue server id: 1

[ 19993][D][ModbusClientRTU.cpp:278] handleConnection(): Data response (7 bytes) received.

[ 19994][D][ModbusClientRTU.cpp:259] handleConnection(): Pulled request from queue server id: 1

[ 20022][D][ModbusClientRTU.cpp:278] handleConnection(): Data response (7 bytes) received.

[ 20022][D][ModbusClientRTU.cpp:259] handleConnection(): Pulled request from queue server id: 1

[ 20050][D][ModbusClientRTU.cpp:278] handleConnection(): Data response (7 bytes) received.

[ 20051][D][ModbusClientRTU.cpp:259] handleConnection(): Pulled request from queue server id: 2

[ 22061][D][ModbusClientRTU.cpp:278] handleConnection(): Error response (1 bytes) received.

[ 22062][D][ModbusClientRTU.cpp:259] handleConnection(): Pulled request from queue server id: 1

[ 22091][D][ModbusClientRTU.cpp:278] handleConnection(): Data response (7 bytes) received.

[ 22091][D][ModbusClientRTU.cpp:259] handleConnection(): Pulled request from queue server id: 1

[ 22119][D][ModbusClientRTU.cpp:278] handleConnection(): Data response (7 bytes) received.

[ 22120][D][ModbusClientRTU.cpp:259] handleConnection(): Pulled request from queue server id: 1

[ 22147][D][ModbusClientRTU.cpp:278] handleConnection(): Data response (7 bytes) received.

[ 22148][D][ModbusClientRTU.cpp:259] handleConnection(): Pulled request from queue server id: 2

[ 24158][D][ModbusClientRTU.cpp:278] handleConnection(): Error response (1 bytes) received.

[ 24159][D][ModbusClientRTU.cpp:259] handleConnection(): Pulled request from queue server id: 1

[ 24188][D][ModbusClientRTU.cpp:278] handleConnection(): Data response (7 bytes) received.

[ 24189][D][ModbusClientRTU.cpp:259] handleConnection(): Pulled request from queue server id: 1

[ 24218][D][ModbusClientRTU.cpp:278] handleConnection(): Data response (7 bytes) received.

[ 24218][D][ModbusClientRTU.cpp:259] handleConnection(): Pulled request from queue server id: 1

[ 24247][D][ModbusClientRTU.cpp:278] handleConnection(): Data response (7 bytes) received.

[ 24247][D][ModbusClientRTU.cpp:259] handleConnection(): Pulled request from queue server id: 2

[ 26260][D][ModbusClientRTU.cpp:278] handleConnection(): Error response (1 bytes) received.

[ 26261][D][ModbusClientRTU.cpp:259] handleConnection(): Pulled request from queue server id: 1

[ 26289][D][ModbusClientRTU.cpp:278] handleConnection(): Data response (7 bytes) received.

[ 26289][D][ModbusClientRTU.cpp:259] handleConnection(): Pulled request from queue server id: 1

[ 26317][D][ModbusClientRTU.cpp:278] handleConnection(): Data response (7 bytes) received.

[ 26318][D][ModbusClientRTU.cpp:259] handleConnection(): Pulled request from queue server id: 1

[ 26345][D][ModbusClientRTU.cpp:278] handleConnection(): Data response (7 bytes) received.

[ 26346][D][ModbusClientRTU.cpp:259] handleConnection(): Pulled request from queue server id: 2

[ 28356][D][ModbusClientRTU.cpp:278] handleConnection(): Error response (1 bytes) received.
```

I-Connect, 2024-02-16:
ow btw, thx for helping and the quick response :-)

Miq1, 2024-02-16:
What you are missing is the "Data response..." line is logged after the quiet time has passed already 😉 
[line 260 in RTUutils.cpp](https://github.com/eModbus/eModbus/blob/efddc7cd0ae6e20872cea37460838bd462d39220/src/RTUutils.cpp#L260)

Miq1, 2024-02-16:
Oh, by the way: if it was a missing quiet time, it would have hit the one-server-polling scenario as well.

Must be a different root cause, but I am eager to find that as well. Too bad I can not simulate that here 😞

I-Connect, 2024-02-16:
Indeed, my logging missed the quiet time,  added two logs:
[image]
(my formatter makes the line nrs to differ from yours...)

So there is a 16 ms wait time (red marked part) here if I interpret it correctly.

At 9600 baud you send 960 chars/sec so that is 1ms per char, if wait time should be 3.5 chars the 16 ms should be more than enough.

[image]

I-Connect, 2024-02-16:
I do not have in depth knowledge of the Modbus timing, maybe there is a difference in quit time when switching between devices? Does a device stop listening when a message is received for a different server id?

I do have a lot of other tasks running (also modbus TCP), but as RTU uses a buffered HW serial and I think there are no task switches possible (no yields or delays) after a request is pulled from the queue, I would expect this to not have any influence. I can see if I can disable some functionality

I-Connect, 2024-02-16:
[image]

hmmm I have the SDM120 and SDM 630, but just the fact that they state such a long silent time for a different device makes me worry as this is not the standard...

Miq1, 2024-02-16:
Yeah, some devices will need a "refresh" time between requests, but that is no part of the standard.

Miq1, 2024-02-16:
The 1.5 interruption detection interval they are stating is non-standard as well... eModbus does not track that at all, btw, a gap may come close to the 3.5c and still will not be detected.

I-Connect, 2024-02-18:
As you state there are more device that need this, would you consider adding a method for an optional extended silent period ("refreshTime") parameter?

[image]

Otherwise I would need to fork your repo as I will need this (and I prefer to use the original one...)

If yes, what would be wise, pass it to RTUutils::receive(...) as a parameter and add it to the timeout interval or add it as a delay at the end of  ModbusClientRTU::handleConnection when a request has been pulled from the queue?

Miq1, 2024-02-18:
That would be another deviation from the standard, but we have other optional ones already. Yeah, will do. Would you mind to test it once it has been added?

Miq1, 2024-02-18:
The ``master`` branch now has a modified ``begin()``, the user defined interval (aka quiet time) is to be given as a new last parameter:
```
void begin(Stream& serial, uint32_t baudRate, int coreID = -1, uint32_t userInterval = 0);
void begin(HardwareSerial& serial, int coreID = -1, uint32_t userInterval = 0);
```
Please note that you will have to provide a core (0, 1 or -1 for the default) to be able to specify another interval.
Please not also that the user interval may not be shorter than the calculated standard interval to not break existing implementations.
A caveat: if you will specify a very long interval that exceeds the time the UART buffers will take to fill up, you will experience data losses with messages longer than a UART buffer can fit!

I-Connect, 2024-02-19:
awesome, thx! Tested with 10 ms and it works, I will leave it running for some time.

Small suggestion, maybe change the parameter name to userIntervalUs (I made the error that I thought it was ms :-) )

> A caveat: if you will specify a very long interval that exceeds the time the UART buffers will take to fill up, you will experience data losses with messages longer than a UART buffer can fit!

Not sure if I understand this, isn't the wait time after a 
 -   request has been pulled from the queue - request sent - answer received   - 
then there is a wait before the next request will be pulled from the queue and sent.
So the serial buffer could not fill up...?

Miq1, 2024-02-19:
You are seeing the client side only. A server needs to maintain the same timings and may miss the next request if the client on the bus will send too early.

I-Connect, 2024-02-20:
I increased the userInterval to 15 ms and apart from an occasional single timeout it runs fine

Miq1, 2024-02-20:
Glad it works for you. Thank you for reporting this.

closed 2024-02-20 by Miq1

I-Connect, 2024-02-20:
thank you very much for the help!