eModbus/eModbus #340: list of edge cases and do not try scenario

sdrshnptl, 2024-03-06:
Hello All!
this is a fantastic library to achieve modbus functionality over serial, WiFi and Ethernet.

The intention of this issue is to have a list of edge cases and esp32 will reset or misbehave in running that been found which are not covered in documentation and it will improve performance.

Thanks!
Sudarshan

Miq1, 2024-03-06:
What comes to my mind first is race conditions. It is in the docs, but maybe slightly hidden:

- all callbacks (data and error handlers for the client and worker functions for the server) are run in a task within the library and may compete with the main task in terms of data access and resources.
- as a side note: these tasks are running with limited (4kB by default) stack space also.

Miq1, 2024-03-06:
Second, do not overrun your servers with requests. Polling servers faster than they can respond will disable your client in the end.

Miq1, 2025-04-10:
If you do short requests - a few bytes only - on RTU addressed to several servers in quick succession that will result in equally short responses, ``ModbusServerRTU`` may be unable to recover from the last message on the bus before the next is already incoming. This typically will lead to ``0xE2 CRC`` errors, because the last response from another server and the next request will be mixed in the UART RX buffer.

To prevent that, use the optional ``userInterval`` parameter in the ``begin()`` call on the client to extend the intra-message interval:
```
void begin(HardwareSerial& serial, int coreID = -1, uint32_t userInterval = 0);
```
The interval is given in µs, the lowest possible value the Modbus standard allows is 1750µs. At higher baud rates this minimum will be hit, so extending it to, say, 3000 will reduce the described effect.

Miq1, 2025-04-10:
The ESP32 core TCP stack does not react well to connections physically broken while active, like switching off servers mid-action or pulling the LAN cable. The connection will still be reported as being active for a long time. This can hang up your ``ModbusClientTCP``.

To prevent that, you may use the ``ModbusClientTCP::closeConnectionOnTimeouts()`` call:
```
  // Set number of timeouts to tolerate before a connection is forcibly closed.
  // 0: never, 1..255: desired number
  // Returns previous value.
  uint8_t closeConnectionOnTimeouts(uint8_t n=3);
```
If the given number of timeouts have been hit in a row, the connection will be closed regardless (and re-opened for the next request in case).