eModbus/eModbus #343: Help to use this library with an anemometer ultrasonic

carminelau, 2024-03-28:
hello everyone,

I use an ESP32 S3, a MAX485 and an ultrasonic anemometer via RS485. I would like to replace the current library with this one but I cannot figure out how to make it work. 

Can anyone help me out ?
[Manuale Anemometro.pdf](https://github.com/eModbus/eModbus/files/14790261/Manuale.Anemometro.pdf)

and this is my pinout:
DI: 6
RO: 7
RE: 4
DE: 5

Miq1, 2024-03-28:
You will not need to use two pins for RE and DE, just connect both to a single GPIO. ``Serial1`` should be set up to use GPIOs 6 and 7. The single pin for RE/DE must be given with the definition call for your ``ModbusClientRTU``. The ``begin`` call then will have ``Serial1`` as parameter and off you go - normally 😁

carminelau, 2024-03-28:
can u send me an example ?

Miq1, 2024-03-28:
Have a look at the [``ModbusClientRTU example in the docs](https://emodbus.github.io/modbusclient) that has some explanations with it.

Miq1, 2024-04-01:
Closing this as it seems to be not relevant any more.

closed 2024-04-01 by Miq1

carminelau, 2024-04-03:
Hi, sorry to reopen this issue, but as I have only just started working with esp32 and RS485 can you write me a complete example ? In the example you tagged me I cannot understand the token what it is

Miq1, 2024-04-03:
I am afraid the new example would not look much different from that I linked in the last post. The documentation you found it in has much more information on the mechanisms working - you will want to invest some time reading and understanding it.

The token is a mark you will define and supply to the call, that you can use to identify the request to a received response. It can be any uint32_t number you like - 1, 4711, whatever. You send it with your request and you will get it back in the response/data/error handler. Imagine you are sending a bunch of requests to several different servers; all responses will end up in your one handler function, so you need to tell for which request the current response was received.