eModbus/eModbus #351: Why using Ethernet repo of Max GErhardt in stead of arduino-libraries?

Bascy, 2024-04-25:
#### Do not use to report bugs!

**Is your feature request related to a problem? Please describe.**
Ethernet fork of Max Gerhardt hasnt been updated, while the arduin-libraries seems te be more alive...
Also I have posted an issue with the arduino-library, for which I don't know if Max Gerhardt will merge it into his repo 

**Describe the solution you'd like**
Replace dependency `https://github.com/maxgerhardt/Ethernet.git` with `https://github.com/arduino-libraries/Ethernet.git`

Miq1, 2024-04-26:
That choice of Max' Ethernet lib was made a few years ago and I honestly never cared about it again since. At that time the Arduino core Ethernet lib was missing a bit to be completely compliant and so I had to use an alternative.
I will check if the core lib now is doing its work and switch over in case.

Thank you for the suggestion.

Miq1, 2024-04-27:
I really tried, but I can not get the code to compile with any ``Ethernet.h`` at all. And the root cause for Max' version seems still to be valid - a mismatch between the ``connect()`` definitions in the Arduino core and the ESP IDF.

Do you have suggestions as how to use the standard Ethernet lib?

Bascy, 2024-04-28:
As it turns out we only use the ModbusClient so we're not affected by the different Ethernet repo you use.
So no prio for this now. if / when we're gonna use ModbusServer functionality I'll look into this again

CLosing the issue

closed 2024-04-28 by Bascy