eModbus/eModbus #359: eModbus and ESPAsyncWebServer crashes

godchadigo, 2024-05-19:
#### Do not use to discuss topics! 
#### Please forgive me for my poor English, I can only present it as a translation.
#### Is it possible that eModbus and otowinter/ESPAsyncWebServer-esphome can't coexist, I just have to send a GET request and ESP32 crashes.
I've tested running eModbus alone or ESPAsyncWebServer alone and it works fine, but when both packages are run together it crashes.

**Describe the bug**
A clear and concise description of what the bug is. Which components of eModbus did you use?
1. #include "ModbusServerTCPasync.h"
**Expected behavior**
A clear and concise description of what you expected to happen.
1. It shouldn't have collapsed.
**To Reproduce**
Steps to reproduce the behavior:
1. init network configure
2. init wifi (client mode)
3. connect wifi
4. use postman GET request.
5. GET "http://192.168.0.184/". then esp32 will crash.
**Example code**
```
//Put code here to reproduce the bug, if possible
#include <Arduino.h>

#include "WiFi.h"
#include <ESPAsyncWebServer.h>
#include "AsyncJson.h"
#include "ArduinoJson.h"

#include "ModbusServerTCPasync.h"

/*Wifi*/
const char *ssid = "WiFi_ESP32";   // 設定一組網路名稱(ssid)
const char *password = "12345678"; // 設定一組網路密碼(pasword)

const char *ClientSsid = "CB_Plus";	
const char *ClientPassword = "wifi password";

/*System*/
bool isServer = false;

/*Network*/
IPAddress local_IP(192, 168, 0, 184); // Set the static IP address to use if the DHCP fails
IPAddress gateway(192, 168, 0, 1);    // Set your network Gateway usually your Router base address
IPAddress subnet(255, 255, 255, 0);   // Set your network Subnet mask
IPAddress primaryDNS(8, 8, 8, 8);     // Optional
IPAddress secondaryDNS(8, 8, 4, 4);   // Optional

AsyncWebServer server(80);

void APWifiInit();
void WifiClientInit();
void RestfulServiceInit();

void setup() {
  if (isServer)
  {
    // AP init
    APWifiInit();
    Serial.println("APWifiInit init");
  }
  else
  {
    // Client init
    WifiClientInit();
    Serial.println("WifiClientInit init");
  }

  // 獲取並打印MAC地址
  String mac = WiFi.macAddress(); // 獲得 MAC 地址
  String message = "ESP32 MAC Address: ";
  message.concat(mac); // 將 MAC 地址添加到消息字符串
  Serial.println(message);

  RestfulServiceInit();
  Serial.println("restful init\n");
}

void loop() {
  // put your main code here, to run repeatedly:
}

void APWifiInit()
{
  WiFi.mode(WIFI_AP);

  WiFi.softAPConfig(local_IP, gateway, subnet);
  WiFi.softAP(ssid, password);
}

void WifiClientInit()
{
  // Configures static IP address
  if (!WiFi.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS))
  {
    Serial.println("STA Failed to configure");
  }

  // Connect to Wi-Fi
  WiFi.begin(ClientSsid, ClientPassword);
  while (WiFi.status() != WL_CONNECTED)
  {

    Serial.print(".");
    gpio_set_level(GPIO_NUM_2, !gpio_get_level(GPIO_NUM_2));
    delay(500);
  }
  
}

void RestfulServiceInit()
{

  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request)
            { request->send(200, "text/plain", "Hello, world!"); });

  // Setup route and handle using query parameter
  // Send a GET request to <IP>/get?message=<message>
  server.on("/get", HTTP_GET, [](AsyncWebServerRequest *request)
            {
        String command;
        String parameter;

        // Check if "command" parameter exists
        if (request->hasParam("command")) {
          command = request->getParam("command")->value();
        } else {
          command = "No command provided";
        }

        // Check if "parameter" parameter exists
        if (request->hasParam("parameter")) {
          parameter = request->getParam("parameter")->value();
        } else {
          parameter = "No parameter provided";
        }

        //response
        JsonDocument res;
        String output;
        if (command == "infomation"){     
          //build json doc
          res["heap"] = ESP.getFreeHeap();
          res["ssid"] = WiFi.SSID();
          //Serialize
          
          serializeJson(res, output);                                        
        }  

        if (command == "led"){
          if (parameter == "0"){
            gpio_set_level(GPIO_NUM_2, 0);
            res["code"] = 200;
            res["status"] = "off";            
            serializeJson(res, output);
          }
          if (parameter == "1"){
            gpio_set_level(GPIO_NUM_2, 1);
            res["code"] = 200;
            res["status"] = "on";            
            serializeJson(res, output);
          }
          if (parameter == "blink"){
            //setBlink(5,2);
            res["code"] = 200;
            res["status"] = "blink";            
            serializeJson(res, output);
          }
        }
       
        //return response
        request->send(200, "application/json", output); });

  server.begin();
}

```

**Additional context**
Add any other context about the problem here.

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d980b  PS      : 0x00060330  A0      : 0x800d7da2  A1      : 0x3ffd0f80  
A2      : 0x003fffff  A3      : 0x3ffd168c  A4      : 0x00000000  A5      : 0x00000001
A6      : 0x00000001  A7      : 0x3ffccc98  A8      : 0x00000000  A9      : 0x3ffccc98
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000002  A13     : 0x3ffccc9a  
A14     : 0x0000000d  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00400003  LBEG    : 0x40089885  LEND    : 0x400898a7  LCOUNT  : 0xffffffff

Backtrace: 0x400d9808:0x3ffd0f80 0x400d7d9f:0x3ffd0fa0 0x400d7ef1:0x3ffd0ff0 0x400d80d9:0x3ffd1040 0x4014094d:0x3ffd1060 0x401409c5:0x3ffd1090 0x401411f6:0x3ffd10b0

ELF file SHA256: 848c1b05c22e2b88

Rebooting...
ets Jul 29 2019 12:21:46

godchadigo, 2024-05-19:
[image]

godchadigo, 2024-05-19:
platformio.ini

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp-wrover-kit]
platform = espressif32
board = esp-wrover-kit
framework = arduino
monitor_speed = 115200

lib_deps = 
    ottowinter/ESPAsyncWebServer-esphome@^3.2.0
    bblanchon/ArduinoJson@^7.0.4
    https://github.com/eModbus/eModbus.git
    SPI
```

Miq1, 2024-05-19:
I am not fluent with the async stuff, so I would like to pull in @bertmelis here, as he has much more experience with it.

Is there a reason why you are preferring the async varieties? You seem to be using an ESP32 that would support the regular TCP client and server as well. I reckon these could coexist with the ESPhome stuff.

Miq1, 2024-05-19:
Oh, BTW: could you install the esp_exception_decoder add-on for PlatformIO? Then you would know where in the code the crash was happening.

godchadigo, 2024-05-19:
> Oh, BTW: could you install the esp_exception_decoder add-on for PlatformIO? Then you would know where in the code the crash was happening.

Backtrace: 0x400d9818:0x3ffd0f50 0x400d7daf:0x3ffd0f70 0x400d7f01:0x3ffd0fc0 0x400d80e9:0x3ffd1010 0x4014095d:0x3ffd1030 0x401409d5:0x3ffd1060 0x40141206:0x3ffd1080
  #0  0x400d9818 in AsyncWebServer::_rewriteRequest(AsyncWebServerRequest*) at .pio/libdeps/esp-wrover-kit/ESPAsyncWebServer-esphome/src/WebServer.cpp:106
  #1  0x400d7daf in AsyncWebServerRequest::_parseLine() at .pio/libdeps/esp-wrover-kit/ESPAsyncWebServer-esphome/src/WebRequest.cpp:569
  #2  0x400d7f01 in AsyncWebServerRequest::_onData(void*, unsigned int) at .pio/libdeps/esp-wrover-kit/ESPAsyncWebServer-esphome/src/WebRequest.cpp:123
  #3  0x400d80e9 in std::_Function_handler<void (void*, AsyncClient*, void*, unsigned int), AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(void*, AsyncClient*, void*, unsigned int)#8}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, std::_Any_data const&, unsigned int&&) at .pio/libdeps/esp-wrover-kit/ESPAsyncWebServer-esphome/src/WebRequest.cpp:76
      (inlined by) _M_invoke at c:\users\chadigo\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:297
  #4  0x4014095d in std::function<void (void*, AsyncClient*, void*, unsigned int)>::operator()(void*, AsyncClient*, void*, unsigned int) const at c:\users\chadigo\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:687
      (inlined by) AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at .pio/libdeps/esp-wrover-kit/AsyncTCP-esphome/src/AsyncTCP.cpp:972
  #5  0x401409d5 in AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) at .pio/libdeps/esp-wrover-kit/AsyncTCP-esphome/src/AsyncTCP.cpp:1292
  #6  0x40141206 in _async_service_task(void*) at .pio/libdeps/esp-wrover-kit/AsyncTCP-esphome/src/AsyncTCP.cpp:164
      (inlined by) _async_service_task at .pio/libdeps/esp-wrover-kit/AsyncTCP-esphome/src/AsyncTCP.cpp:199

godchadigo, 2024-05-19:
> Backtrace: 0x400d9818:0x3ffd0f50 0x400d7daf:0x3ffd0f70 0x400d7f01:0x3ffd0fc0 0x400d80e9:0x3ffd1010 0x4014095d:0x3ffd1030 0x401409d5:0x3ffd1060 0x40141206:0x3ffd1080 #0 0x400d9818 in [...]

[image]

Miq1, 2024-05-19:
Thank you.

Interestingly you just included the eModbus ``ModbusServerTCPasync.h`` without even defining any object from it, so the issue must sit in the general code somewhere.

@bertmelis can you please have a knowledged look?

bertmelis, 2024-05-19:
esphome uses a custom AsyncTCP library. When you include eModbus there might be a conflict with different versions.

Another possible issue is the use of lambdas. The function object is destroyed after exiting `void RestfulServiceInit()` so the webserver calls a non existing function. Try with regular functions.

godchadigo, 2024-05-20:
@bermeils I see. Thank you for your answer.

closed 2024-05-20 by godchadigo