eModbus/eModbus #360: REDEPIN not toggling

vaclavvava, 2024-05-26:
Bert Melis esp32ModbusRTU (esp32ModbusRTU-master) + RTU04example
ESP32 DEVKIT (=MH-ET LIVE ESP32 Development Board)
REDEPIN stays in LOW state. I have a TTL/RS485 board with R/D pin.
The esp32ModbusRTU-master library works correctly with the example for SDM630, the RTS pin changes state, and the data is read correctly from Modbus. I would like to use the newer eModbus library, but I am not a programmer and cannot detect the reason why the Modbus read/write control pin is not changing state.
Thank you for your help.

```

vaclavvava, 2024-05-26:
I puzzled over it for five days, then submitted the issue, and in another 5 minutes I found a solution:
there's a bug in the sample example. There are wrong parameters in brackets, the Serial2 one is wrong and shouldn't be there.
ModbusClientRTU MB(Serial2, REDEPIN); //wrong entry
ModbusClientRTU MB(REDEPIN); //correct entry

vaclavvava, 2024-05-26:
thanks for a great library

closed 2024-05-26 by vaclavvava

Miq1, 2024-05-26:
Ough. Sorry, my bad. The example still has the syntax from before the modified UART handling and needs to be corrected.
Thank you for pointing me to it! 👍🏻

Miq1, 2024-05-26:
I have to come back to it again; I looked into the RTU04example code and only found the correct line?!? Where did you find the faulty one exactly?

vaclavvava, 2024-05-26:
I made a mistake in first line my first comment. I edited it. 
This does not apply to the library example esp32ModbusRTU-master, but
eModbus-master\examples\RTU04example\.
There is on line 33:
ModbusClientRTU MB(Serial2, REDEPIN);
For me it only works like this: ModbusClientRTU MB(REDEPIN);

This corresponds to information from 
https://emodbus.github.io/modbusclient-rtu-api
where is:
ModbusClientRTU(),
**ModbusClientRTU(int8_t rtsPin) and**
ModbusClientRTU(int8_t rtsPin, uint16_t queueLimit)
These are the constructor variants for an instance of the ModbusClientRTU type.

vaclavvava, 2024-05-26:
Hi,

I made a mistake in first line my first comment. I edited it.
This does not apply to the library example esp32ModbusRTU-master, but
eModbus-master\examples\RTU04example.
There is on line 33:
ModbusClientRTU MB(Serial2, REDEPIN);
For me it only works like this: ModbusClientRTU MB(REDEPIN);

This corresponds to information from
https://emodbus.github.io/modbusclient-rtu-api
(https://emodbus.github.io/modbusclient-rtu-api)
where is:
ModbusClientRTU(),
ModbusClientRTU(int8_t rtsPin) and
ModbusClientRTU(int8_t rtsPin, uint16_t queueLimit)
These are the constructor variants for an instance of the ModbusClientRTU 
type.

---------- Původní e-mail ----------

Komu: eModbus/eModbus ***@***.***>
Kopie: vaclavvava ***@***.***>, State change ***@***.***
github.com>
Datum: 26. 5. 2024 18:09:29
"

(https://github.com/eModbus/eModbus/issues/360#issuecomment-2132273016), or 
unsubscribe
"

Miq1, 2024-05-26:
Please have a look [here at the RTU04example](https://github.com/eModbus/eModbus/blob/master/examples/RTU04example/main.cpp). 
line 33 is ``ModbusClientRTU MB(REDEPIN);``, so you may have taken some older source?

vaclavvava, 2024-05-26:
I'm very sorry, I have rev=1.4.1, current version=1.7.2. . 
Here is the crux of the matter... 

---------- Původní e-mail ----------

Datum: 26. 5. 2024 21:57:43
"

(https://github.com/eModbus/eModbus/blob/master/examples/RTU04example/main.cpp)

Miq1, 2024-05-26:
No worries, good that you found it.