eModbus/eModbus #363: Not compiling for ESP32-C3

BorisBrock, 2024-07-05:
I just created a blank project with Platformio for the ESP32-C3 mcu. Here's the platformio.ini:
```
[env:lolin_c3_mini]
platform = espressif32
board = lolin_c3_mini
framework = arduino
lib_deps =
  miq19/eModbus @ ^1.7.2
```
When compiling the project, there's tons of errors (see below). Removing the eModbus library from the project results in a successful build.
So it seems the eModbus library is not compatible with the ESP32-C3. Can this be fixed?

Here are the compiler errors:

> Processing lolin_c3_mini (platform: espressif32; board: lolin_c3_mini; framework: arduino)
> ----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
> Verbose mode can be enabled via `-v, --verbose` option
> CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/lolin_c3_mini.html
> PLATFORM: Espressif 32 (6.7.0) > WEMOS LOLIN C3 Mini
> HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
> DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
> PACKAGES:
>  - framework-arduinoespressif32 @ 3.20016.0 (2.0.16)
>  - tool-esptoolpy @ 1.40501.0 (4.5.1)
>  - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
> LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
> LDF Modes: Finder ~ chain, Compatibility ~ soft
> Found 36 compatible libraries
> Scanning dependencies...
> Dependency Graph
> |-- eModbus @ 1.7.2
> Building in release mode
> Archiving .pio\build\lolin_c3_mini\libd34\libAsyncTCP.a
> Compiling .pio\build\lolin_c3_mini\lib2fc\Ethernet\Dhcp.cpp.o
> Compiling .pio\build\lolin_c3_mini\lib2fc\Ethernet\Dns.cpp.o
> Compiling .pio\build\lolin_c3_mini\lib2fc\Ethernet\Ethernet.cpp.o
> Compiling .pio\build\lolin_c3_mini\lib2fc\Ethernet\EthernetClient.cpp.o
> Compiling .pio\build\lolin_c3_mini\lib2fc\Ethernet\EthernetServer.cpp.o
> Compiling .pio\build\lolin_c3_mini\lib2fc\Ethernet\EthernetUdp.cpp.o
> Compiling .pio\build\lolin_c3_mini\lib2fc\Ethernet\socket.cpp.o
> Compiling .pio\build\lolin_c3_mini\lib2fc\Ethernet\utility\w5100.cpp.o
> Archiving .pio\build\lolin_c3_mini\lib0f6\libeModbus.a
> Compiling .pio\build\lolin_c3_mini\FrameworkArduino\USBMSC.cpp.o
> Compiling .pio\build\lolin_c3_mini\FrameworkArduino\WMath.cpp.o
> Compiling .pio\build\lolin_c3_mini\FrameworkArduino\WString.cpp.o
> Compiling .pio\build\lolin_c3_mini\FrameworkArduino\base64.cpp.o
> In file included from .pio/libdeps/lolin_c3_mini/Ethernet/src/Dhcp.cpp:7:
> .pio/libdeps/lolin_c3_mini/Ethernet/src/utility/w5100.h:18:10: fatal error: SPI.h: No such file or directory
> 
> *************************************************************
> * Looking for SPI.h dependency? Check our library registry!
> *
> * CLI  > platformio lib search "header:SPI.h"
> * Web  > https://registry.platformio.org/search?q=header:SPI.h
> *
> *************************************************************
> 
>  #include <SPI.h>
>           ^~~~Compiling .pio\build\lolin_c3_mini\FrameworkArduino\cbuf.cpp.o
> ~~~
> compilation terminated.
> In file included from .pio/libdeps/lolin_c3_mini/Ethernet/src/EthernetClient.cpp:24:
> .pio/libdeps/lolin_c3_mini/Ethernet/src/utility/w5100.h:18:10: fatal error: SPI.h: No such file or directory
> 
> *************************************************************
> * Looking for SPI.h dependency? Check our library registry!
> *
> * CLI  > platformio lib search "header:SPI.h"
> * Web  > https://registry.platformio.org/search?q=header:SPI.h
> *
> *************************************************************
> 
>  #include <SPI.h>
>           ^~~~~~~
> compilation terminated.
> In file included from .pio/libdeps/lolin_c3_mini/Ethernet/src/Dns.cpp:8:
> .pio/libdeps/lolin_c3_mini/Ethernet/src/utility/w5100.h:18:10: fatal error: SPI.h: No such file or directory
> 
> *************************************************************
> * Looking for SPI.h dependency? Check our library registry!
> *
> * CLI  > platformio lib search "header:SPI.h"
> * Web  > https://registry.platformio.org/search?q=header:SPI.h
> *
> *************************************************************
> 
>  #include <SPI.h>
>           ^~~~~~~
> In file included from .pio/libdeps/lolin_c3_mini/Ethernet/src/Ethernet.cpp:23:
> .pio/libdeps/lolin_c3_mini/Ethernet/src/utility/w5100.h:18:10: fatal error: SPI.h: No such file or directory
> 
> *************************************************************
> * Looking for SPI.h dependency? Check our library registry!
> *
> * CLI  > platformio lib search "header:SPI.h"
> * Web  > https://registry.platformio.org/search?q=header:SPI.h
> *
> *************************************************************
> 
>  #include <SPI.h>
>           ^~~~~~~
> compilation terminated.
> compilation terminated.
> Compiling .pio\build\lolin_c3_mini\FrameworkArduino\esp32-hal-adc.c.o
> *** [.pio\build\lolin_c3_mini\lib2fc\Ethernet\Dhcp.cpp.o] Error 1
> *** [.pio\build\lolin_c3_mini\lib2fc\Ethernet\Dns.cpp.o] Error 1
> *** [.pio\build\lolin_c3_mini\lib2fc\Ethernet\Ethernet.cpp.o] Error 1
> *** [.pio\build\lolin_c3_mini\lib2fc\Ethernet\EthernetClient.cpp.o] Error 1
> In file included from .pio/libdeps/lolin_c3_mini/Ethernet/src/EthernetServer.cpp:23:
> .pio/libdeps/lolin_c3_mini/Ethernet/src/utility/w5100.h:18:10: fatal error: SPI.h: No such file or directory
> 
> *************************************************************
> * Looking for SPI.h dependency? Check our library registry!
> *
> * CLI  > platformio lib search "header:SPI.h"
> * Web  > https://registry.platformio.org/search?q=header:SPI.h
> *
> *************************************************************
> 
>  #include <SPI.h>
>           ^~~~~~~
> compilation terminated.
> *** [.pio\build\lolin_c3_mini\lib2fc\Ethernet\EthernetServer.cpp.o] Error 1
> In file included from .pio/libdeps/lolin_c3_mini/Ethernet/src/utility/w5100.cpp:13:
> .pio/libdeps/lolin_c3_mini/Ethernet/src/utility/w5100.h:18:10: fatal error: SPI.h: No such file or directory
> 
> *************************************************************
> * Looking for SPI.h dependency? Check our library registry!
> *
> * CLI  > platformio lib search "header:SPI.h"
> * Web  > https://registry.platformio.org/search?q=header:SPI.h
> *
> *************************************************************
> 
>  #include <SPI.h>
>           ^~~~~~~
> compilation terminated.
> *** [.pio\build\lolin_c3_mini\lib2fc\Ethernet\utility\w5100.cpp.o] Error 1
> In file included from .pio/libdeps/lolin_c3_mini/Ethernet/src/socket.cpp:23:
> .pio/libdeps/lolin_c3_mini/Ethernet/src/utility/w5100.h:18:10: fatal error: SPI.h: No such file or directory
> 
> *************************************************************
> * Looking for SPI.h dependency? Check our library registry!
> *
> * CLI  > platformio lib search "header:SPI.h"
> * Web  > https://registry.platformio.org/search?q=header:SPI.h
> *
> *************************************************************
> 
>  #include <SPI.h>
>           ^~~~~~~
> compilation terminated.
> *** [.pio\build\lolin_c3_mini\lib2fc\Ethernet\socket.cpp.o] Error 1
> In file included from .pio/libdeps/lolin_c3_mini/Ethernet/src/EthernetUdp.cpp:32:
> .pio/libdeps/lolin_c3_mini/Ethernet/src/utility/w5100.h:18:10: fatal error: SPI.h: No such file or directory
> 
> *************************************************************
> * Looking for SPI.h dependency? Check our library registry!
> *
> * CLI  > platformio lib search "header:SPI.h"
> * Web  > https://registry.platformio.org/search?q=header:SPI.h
> *
> *************************************************************
> 
>  #include <SPI.h>
>           ^~~~~~~
> compilation terminated.
> *** [.pio\build\lolin_c3_mini\lib2fc\Ethernet\EthernetUdp.cpp.o] Error 1
> =============================================================================================================== [FAILED] Took 5.63 seconds ===============================================================================================================

Miq1, 2024-07-05:
Would you mind to try again with the "official" 1.7.2 release? Your ``lib-deps`` should have a plain ``eModbus`` in it.

BorisBrock, 2024-07-05:
`lib_ldf_mode = deep+` did the trick. Now it compiles fine. Problem solved 👍 

One more question though:
Will things like the **Modbus bridge** work well on a single core ESP32 (like the C3)? Or is a dual core ESP required?

closed 2024-07-05 by BorisBrock

Miq1, 2024-07-05:
The number of cores is irrelevant as long as we have FreeRTOS tasks. Even the plain client and server varieties are using two tasks at least on the ESP32. On the basic ESP32 (like the WROOM or WROVER) the two cores may be used for load balancing, but are not required either.