eModbus/eModbus #370: at each request, it prints to the monitor of port 1

yuriy23, 2024-08-25:
at each request, it prints to the monitor of port 1. what is this? and how to disable it?
[https://youtu.be/7XZPfmkFHxs?si=e9KptS1P_wW5jK_E](url)

Miq1, 2024-08-25:
Can you please post the monitor output here? The Youtube- link is not suitable to debug something.

yuriy23, 2024-08-25:
when the modbus is running, it simply prints 1 to the port monitor.

Miq1, 2024-08-25:
Without any code, log list or more detailed descriptions I can not make heads or tails from it. The library will not print anything if not told to do so.

Miq1, 2024-08-26:
@yuriy23 Any progress?

yuriy23, 2024-08-28:
Good afternoon, I was busy. I was setting up my smd mounting line... I reassembled the project and everything fell into place, no longer prints the number 1. I still don't u
[image]
nderstand the reason. But thanks and the topic can be closed. P.S Sorry for my English

Miq1, 2024-08-28:
Thanks for the feedback!

closed 2024-08-28 by Miq1