eModbus/eModbus #375: Creating a ModbusClientRTU instance with no arguments throws error

mace-de, 2024-09-30:
#### Do not use to discuss topics! 

**Describe the bug**
Creating a ModbusClientRTU instance with no arguments, as shown in the example for auto RTS-Mode is not possible and throws an error. When created with -1 as argument, as the default value of rtsPin is, than it's ok.

**To Reproduce**
Use ESP32 Project in VSC/Platformio

**Example code**
```
#include <ModbusClientRTU.h>

ModbusClientRTU mymodbus();
void handleData(ModbusMessage msg, uint32_t token) 
{

}

void setup() {
  Wire.setPins(19,21);
  RTUutils::prepareHardwareSerial(Serial2);
  Serial2.begin(9600,SERIAL_8N1);
  mymodbus.onDataHandler(&handleData);
  Serial.begin(115200);
```
**Additional context**
[image]

Miq1, 2024-09-30:
Strange... The header file has
```
// Constructor takes an optional DE/RE pin and queue limit
  explicit ModbusClientRTU(int8_t rtsPin = -1, uint16_t queueLimit = 100);
```
and the example does compile here without an error. Is ``mymodbus`` that you are using defined anywhere else in your code? The error message is stating a ``ModbusClientRTU (*)()`` that is no constructor, but a function returning a pointer to a ModbusClientRTU.

mace-de, 2024-09-30:
No, mymodbus is just used here. It dosn't mater what name you choose.
Without the brackets it's working too.

Miq1, 2024-09-30:
So I consider the case closed?

mace-de, 2024-09-30:
Think so. Maybe placing a hint somewhere in the doc could be helpful.

Miq1, 2024-09-30:
> Think so. Maybe placing a hint somewhere in the doc could be helpful.

This is standard C++ behaviour AFAIK.

closed 2024-09-30 by Miq1