eModbus/eModbus #390: Dependency Issues ESP32Async/AsyncTCP + me-no-dev/AsyncTCP

yanod-aut, 2025-02-21:
Hi,

im working on a project where i have to use the newer ESP32Async/AsyncTCP(https://github.com/ESP32Async/AsyncTCP).

This introduces build errors - because of the same lib installed twice (me-no-dev/AsyncTCP requirement in the library.json)
https://github.com/eModbus/eModbus/blob/331b9e647a2664a31bc0d998824a9ad6ec806a05/library.json#L26-L32

found my solution by replacing the part in library.json - 
maybe you could look into if a swap to the new fork may be an option ?
```
    {
      "owner": "ESP32Async",
      "name": "AsyncTCP",
      "version": "*",
      "platforms": ["espressif32"]
    },
```

thanks.

bertmelis, 2025-02-21:
Yes. I'll update the dependencies.

Miq1, 2025-02-21:
> Yes. I'll update the dependencies.

Is that an opportunity to move ahead with the ESP8266 dependencies as well?

bertmelis, 2025-02-21:
Both ESP32 and ESP8266 are in the "async" organization so both dependencies will be updated. The main async problem -the hardcoded timeout- is not addressed at the moment though.

Miq1, 2025-02-21:
I see. I suppose as soon as there is a fix it will be incorporated in the (now updated) dependecies?

hagre, 2025-09-21:
possible workaround to compile until fixed:  

in the platformio.ini

lib_deps = 
    miq19/eModbus@^1.7.4
    ESP32Async/AsyncTCP

lib_ignore = 
	ESPAsyncTCP