eModbus/eModbus #393: ESP32 Compilation fails. debug.h and interrupts.h missing

dzalf, 2025-03-10:
Please bear with me. This is my first experience with Modbus and this library and I am totally lost

As described, the compiler throws a bunch of errors, some of them pertaining to missing headers:

[image]

[image]

[image]

I am working on a wrapper Class that integrates a `ModbusServerRTU` object pointer. The implementation is loosely based on the example code `RTUserver`

`wrapper.h`
```cpp
public:
    /**
     * @brief Constructor for rDAQ_Controller.
     * @param wireObj Pointer to the Wire object for I2C communication.
     */
    rDAQ_Controller(TwoWire *wireObj);

    /**
     * @brief Initialize all connected devices.
     */
    void initDevices();

    /**
     * @brief Start the Modbus RTU server.
     * @param serial Reference to the hardware serial interface for RS485 communication.
     * @param baudRate Baud rate for RS485 communication.
     * @param rtsPin GPIO pin for RTS (optional).
     */
    void startModbusServer(HardwareSerial &serial, uint32_t baudRate, int rtsPin = -1);

    /**
     * @brief Stop the Modbus RTU server.
     */
    void stopModbusServer();

    /**
     * @brief Register a worker for a specific Modbus function code.
     * @param serverID Modbus server ID for which the worker is registered.
     * @param functionCode Modbus function code to handle (e.g., READ_HOLD_REGISTER).
     * @param worker Callback function to process the requests.
     */
    void registerModbusWorker(uint8_t serverID, uint8_t functionCode, ModbusMessage (*worker)(ModbusMessage));
...
private:

ModbusServerRTU *_modbusRTU;       ///< ModbusServerRTU instance for RS485 communication.
```

`wrapper.cpp`

```cpp

void wrapper::startModbusServer(HardwareSerial &serial, uint32_t baudRate, int rtsPin)
{
    // Create the Modbus server instance with a 2000ms timeout
    if (!_modbusRTU)
    {
        _modbusRTU = new ModbusServerRTU(2000, rtsPin); // Timeout is 2000ms
    }

    // Prepare the hardware serial for Modbus
    RTUutils::prepareHardwareSerial(serial);

    // Begin the Modbus server with the given serial interface and baud rate
    _modbusRTU->begin(serial);
}

void wrapper::stopModbusServer()
{
    // End the Modbus server and free memory
    if (_modbusRTU)
    {
        _modbusRTU->end();
        delete _modbusRTU;
        _modbusRTU = nullptr;
    }
}

void wrapper::registerModbusWorker(uint8_t serverID, uint8_t functionCode, ModbusMessage (*worker)(ModbusMessage))
{
    if (_modbusRTU)
    {
        // Register a worker function for the given function code
        _modbusRTU->registerWorker(serverID, functionCode, worker);
    }
}
```

Regardless of my implementations, the error is related to the headers.

Any help will be very much appreciated

:beers:

Miq1, 2025-03-10:
This looks like some basic problem in the core libraries. Can you post your complete platformio.ini here and tell us what type of ESP32 you are going to use?

Miq1, 2025-03-10:
BTW, instead of writing ayour own function signature for the ``registerWorker`` call you could use the predefined ``MBSworker`` type like:
```
void wrapper::registerModbusWorker(uint8_t serverID, uint8_t functionCode, MBSworker worker);
```
This is a ``std::function`` type signature that is more versatile than a plain pointer-to-function and will follow all changes the eModbus API may undergo.

dzalf, 2025-03-11:
> This looks like some basic problem in the core libraries. Can you post your complete platformio.ini here and tell us what type of ESP32 you are going to use?

Sure, this is the ini file:

``ini
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino

board_upload.flash_size = 16MB
board_build.f_cpu = 240000000L

build_flags = 
	-I/lib/modbus-esp8266/src
        -DBOARD_HAS_PSRAM
	-O3

lib_deps = adafruit/Adafruit BusIO@^1.17.0
lib_ldf_mode = deep+
```

My project will use an ESP32-WROVER-IE N16R8

dzalf, 2025-03-11:
> BTW, instead of writing ayour own function signature for the `registerWorker` call you could use the predefined `MBSworker` type like:
>
> ```
> void wrapper::registerModbusWorker(uint8_t serverID, [...]

Thanks for the pointer. I am still learning and must confess that I should study and read the entire implementation BEFORE coding unnecessary things.

First, I would like to solve the compilation issues I am facing, then I will refine my wrapper code.

🍻

Miq1, 2025-03-11:
> 
> Sure, this is the ini file:
> 
> [env:esp32dev]
> platform = espressif32
> board = esp32dev
> framework = arduino
> 
> board_upload.flash_size = 16MB
> board_build.f_cpu = 240000000L
> 
> build_flags = 
> 	-I/lib/modbus-esp8266/src
>         -DBOARD_HAS_PSRAM
> 	-O3
> 
> lib_deps = adafruit/Adafruit BusIO@^1.17.0
> lib_ldf_mode = deep+
> 
> My project will use an ESP32-WROVER-IE N16R8

Two things: 
- the ``-I/lib/modbus-esp8266/src`` line looks wrong. What is it supposed to do? You are using an ESP32 instead and the eModbus library of course.
- (may be related) in the ``lib_deps`` section I am missing the eModbus dependency

It may be you have put your fork of eModbus in said directory to modify it, but that will prevent Platformio from resolving further dependencies.

I do not know what you want to achieve, but my approach for a wrapper would rather be to derive it from ``ModbusServerRTU`` and overwrite just the methods that shall behave differently.

Miq1, 2025-03-14:
Seems to have been resolved otherwise.

closed 2025-03-14 by Miq1

dzalf, 2025-03-16:
Dear @Miq1 

Apologies for not getting back to you on time. I am in the midst of a big wrapper Class development and, unfortunately, I lost track of this library evaluation code and I cannot get back to you with further tests at the moment. I am still learning and experimenting with RS485 and Modbus.

It is fine for now if we keep this "issue" closed. If I experience the same issue or I have other questions, I will get back to you

Thanks for your time

BW

dzalf

🍻