eModbus/eModbus #396: eModbus (TCP) not working on a ESP32C3

Hans-O007, 2025-03-14:
I'm trying to compile a ModbusTCP Client (ModbusClientTCPasync.h) application on a ESP32C3.
While it working fine on an ESP32VROOM, it does not compile for a ESP32C3 super mini board.
The cause is the lack of an Serial2 port on an ESP32C3 board.
Generating following problem...

...\Arduino\libraries\eModbus-master\src\RTUutils.cpp: In static member function 'static int RTUutils::UARTinit(HardwareSerial&, int)':
...\Arduino\libraries\eModbus-master\src\RTUutils.cpp:131:17: error: comparison between distinct pointer types 'HardwareSerial*' and 'HWCDC*' lacks a cast
  131 |     if (&serial == &Serial) {
..\Arduino\libraries\eModbus-master\src\RTUutils.cpp:139:25: error: 'Serial2' was not declared in this scope; did you mean 'Serial1'?
  139 |         if (&serial == &Serial2) {
      |                         ^~~~~~~
      |                         Serial1
...\Arduino\libraries\eModbus-master\src\RTUutils.cpp:141:19: error: 'UART2' was not declared in this scope; did you mean 'UART1'?
  141 |           uart = &UART2;
      |                   ^~~~~
      |                   UART1

As no RTU is connected, and I'm only working over IP, is their a way to avoid the use of Serial2?

Miq1, 2025-03-14:
You seem to be using a rather outdated version of eModbus. Please fetch a recent release or the Master branch from here and try again.

Hans-O007, 2025-03-14:
Many thanks for your quick reply.
It looks like it was an older version.
I removed it and reinstalled the one of this site.
It made a simplified version which compiles good now.
I get an E6 error when executing but that's another problem ;).
Again, many thanks
Hans

closed 2025-03-14 by Miq1