eModbus/eModbus #399: TCP RTU Bridge: race condition with multiple clients

zivillian, 2025-03-21:
**Describe the bug**
I'm connecting with two polling clients ([evcc ](https://github.com/evcc-io/evcc) every 10s & [nodered](https://flows.nodered.org/node/node-red-contrib-modbus) every 30s) to a TCP/RTU Bridge ([heidelberg-1p3p](https://github.com/zivillian/heidelberg-1p3p/blob/34e28217fcdf96d09a6a47eb186e4d1c82ead7ea/src/switch.cpp#L31-L33)). Every now and then I get the following error message in evcc:

>heartbeat: modbus: response data size '28' does not match request quantity '1'

The response data size (28) matches the size of the requests I'm issuing from nodered, so it looks like one client receives a response from a different client.

If I'm correct the library is using [millis()](https://github.com/eModbus/eModbus/blob/4dbf670ca1f4c60cc49d8fd1f5679f2aa5ee1aec/src/ModbusBridgeTemp.h#L252-L254) to assign a unique identifiert to the bridge requests. After a quick search, it looks like `millis()` only has a resolution of 10ms, but even with 1ms resolution it may hit a collission.

**Expected behavior**
Clients only receive their own response.

**To Reproduce**
I was able to easily reproduce this with a small c# program running two clients with different request sizes in a loop (~15% collisions).

Miq1, 2025-03-21:
Brilliant find! 💯 

I will implement a better scheme to tell the responses from another.

zivillian, 2025-03-25:
looks good - I've updated my project and my repo is no longer able to trigger the race

Miq1, 2025-03-26:
👍🏻

closed 2025-03-26 by Miq1