eModbus/eModbus #404: The clearQueue() operation does not actually free up space and causes a memory leak

mifarelight, 2025-04-15:
#### Do not use to discuss topics! 

**Describe the bug**
I am still testing the application performance after the Modbus-TCP server is disconnected from the network.
With the latest version, the application can quickly restore the connection when the server is only disconnected for a short time.

However, if the Modbus-TCP server is disconnected for a long time, if I try to clear the queue after accumulating a certain number of messages in the queue, I can observe that the Free Heap decreases with each clear operation. After multiple clears, the system crashes.
If I remove the clear queue operation, the crash does not occur..

**Expected behavior**
clearQueue() proper release queue and free memory space.

**To Reproduce**
Steps to reproduce the behavior:
Running the code, turn off the Modbus-TCP server for a long time.

**Example code**
```
void modbusTask(void * parameter)
{
  BaseType_t xWasDelayed;
  TickType_t xLastTickTime;
  Error err;

  MB.onDataHandler(&handleData);
  MB.onErrorHandler(&handleError);
  MB.setTimeout();
  MB.begin();
  MB.setTarget(ip, port);
  MB.closeConnectionOnTimeouts();

  xLastTickTime = xTaskGetTickCount();

  for (;;)
  {
    if (MB.pendingRequests() < 32)
    {
      // issue an addRequest(), (addRequest every 1 second in a infinite loop)
      err = MB.addRequest(MB_TOKEN_TIMER_GET, 4, READ_HOLD_REGISTER, 0, 10);
      if (err != SUCCESS)
      {
        ModbusError e(err);
      
        Serial.printf("ERROR: ModBus error request %02X - %s.\n", (int)e, (const char *)e);
      }
    }
    else
    {
      // lost connection too long
      MB.clearQueue();  // clear queue decrease Free Heap / system crashed after multiple clears
      Serial.printf("Try to clear ModBus request queue
    }
    

    xWasDelayed = xTaskDelayUntil(&xLastTickTime, 1000);
  }
}
```

**Additional context**
If I comment "MB.clearQueue();", system won't creash

Miq1, 2025-04-15:
Confirmed, thanks for the report. Found the reason (``RequestEntry `` instances are not individually deleted, just the queue) and will fix it soon.

mifarelight, 2025-04-16:
I just tested it and the memory leak problem has been solved.

Thank you for your support.