eModbus/eModbus #406: Explicitly specifying wired network communication failed

mifarelight, 2025-04-21:
My hardware is ESP32+W5500, so it can communicate via Wi-Fi or wired network.

I used the sample code to communicate with MODBUS-server (PLC device) before, and the sending and receiving with the server worked. Now I want to specify to communicate via wired network (W5500 interface), so I modified below codes and keep the rest untouched:
```
// #include <WiFiClient.h>
#include <Ethernet.h>
```
```
// WiFiClient theClient;
EthernetClient theClient;
```
However, the communication is no longer successful.

Because I set up another Web Server task on ESP32, I can access the web server by the IP allocated (wired network), so I think the wired network is normal at this time.

Is there anything else I should modify, or what test step should I do to find out the problem?

Miq1, 2025-04-21:
That amazes me - it should be working exactly as you coded it. Do you get any error messages? Can you provide a VERBOSE log of the first seconds after booting?

mifarelight, 2025-04-21:
I don't know if these two logs can provide more useful information.

When using "WiFiClient", it failed to connect at first, but then connected successfully.

[modbus.wifi.log](https://github.com/user-attachments/files/19833466/modbus.wifi.log)

When using "EthernetClient", the connection always failed.

[modbus.eth.log](https://github.com/user-attachments/files/19833475/modbus.eth.log)

Miq1, 2025-04-21:
Can you try to extend the ``delay(1)`` on line 280 in ``ModbusClientTCP.cpp`` to - let's say - 50ms? I normally would assume the connection has been made after the ``connect()`` call returned, but who knows?

Miq1, 2025-04-21:
Another variation on that  idea: try using the inter-request interval parameter: [see the docs for it](https://emodbus.github.io/modbusclient-tcp-api#bool-settargetipaddress-host-uint16_t-port--uint32_t-timeout--uint32_t-interval) to give the connection some slack.

Miq1, 2025-04-21:
The W5500 has an internal limit for the number of concurrent connections (6, IIRC). I never found any rule describing when a connection slot can be reused again. Could it be you are hitting that limit with your parallel services running?

mifarelight, 2025-04-21:
Thanks for your fast reply.

I chose to roll back and give it a try:
1. 
```
#include <WiFiClient.h>
//#include <Ethernet.h>
```

```
WiFiClient theClient;
//EthernetClient theClient;
```

2. change back to `delay(1)` on line 280 in `ModbusClientTCP.cpp`

3.  Wi-Fi credentials modified, so Wi-Fi still no way to connect. only W5500

**eModbus works fine as before!**
**Conclusion: the eModbus can work on the W5500 without explicitly specifying.**

---

My problem now should be how to get the MQTT task to work on the W5500. So sad...

closed 2025-04-22 by mifarelight