eModbus/eModbus #407: Support > for ASCII

zivillian, 2025-04-21:
**Is your feature request related to a problem? Please describe.**
[ABL eMH1 uses Modbus ASCII](https://www.ablmobility.de/global/downloads/anleitungen/emh1/Schnittstellenbeschreibung_Modbus-ASCII.pdf?m=1644426447&), but for responses instead of a colon it sends a `>`. [evcc](https://github.com/evcc-io/evcc) supports this Wallbox over Modbus ASCII and is using [grid-x/modbus](https://github.com/grid-x/modbus). They have [explicit support](https://github.com/grid-x/modbus/blob/master/asciiclient.go#L22-L23) for `>`:
```go
// Modbus ASCII defines ':' but in the field often '>' is seen.
var asciiStart = []string{":", ">"}
```

**Describe the solution you'd like**
Also support `>` in replies. If I'm correct it should be sufficient to define it in [RTUUtils::ASCIIread](https://github.com/eModbus/eModbus/blob/a0bb989d9feb34d5cf30d4590ad1a8d2dca99add/src/RTUutils.cpp#L451):
```patch
    /* 30-37 */    0,    1,    2,    3,    4,    5,    6,    7,  // digits 0-7
-   /* 38-3F */    8,    9, 0xF0, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,  // digits 8 + 9, :
+   /* 38-3F */    8,    9, 0xF0, 0xFF, 0xFF, 0xFF, 0xF0, 0xFF,  // digits 8 + 9, : >
    /* 40-47 */ 0xFF,   10,   11,   12,   13,   14,   15, 0xFF,  // digits A-F
```

**Describe alternatives you've considered**
Patch my own fork or do not support ABL eMH1.

**Additional context**
Someone is [trying to integrate an ABL eMH1 into my project](https://github.com/zivillian/heidelberg-1p3p/issues/11), which is using eModbus.

Miq1, 2025-04-22:
Ich könnte mir einen zusätzlichen Call vorstellen, mit dem man das Zeichen explizit freischalten kann. Damit würde im Normalzustand der Standard gelten und man würde das "Fremdzeichen" nur bewusst aktivieren.

Passt das für euch?

aterjung, 2025-04-22:
Hallo, bei mir läuft der Code gerade nur mit der Modifikation im ASCIIread Array. Ich hätte aber auch nichts gegen einen Funktionsaufruf, um das einzuschalten. Siehst du denn einen Nachteil darin, > als Startzeichen generell zu erlauben?

Miq1, 2025-04-22:
Das ist was dogmatisches 😁 Ich möchte im Normalzustand so dicht wie möglich am Standard bleiben.

Miq1, 2025-04-22:
Was willst du denn noch alles mitschicken? Hat dann nicht mehr viel mit Modbus ASCII zu tun...

aterjung, 2025-04-22:
so war das nicht gemeint. Mir reicht das eine Zeichen vollkommen aus. ;-) Ich hatte nur gedacht, das ist dann etwas universeller.

Miq1, 2025-04-22:
Ich verstehe ja eigentlich sowieso nicht, warum manche Hersteller von Standard abweichen müssen. Zumal die Nutzung des '>' ja identisch zu der von ':' im Standard ist. Derlei Possen gibt es beim "echten" binären Protokoll auch gelegentlich. Mag sein, dass die damit die Nutzung mit Hardware anderer Hersteller erschweren wollen.

Ich muss mal sehen, ob ich diese Woche dazu komme, das einzubauen; du kommst ja erstmal mit der gepatchten Tabelle klar.

aterjung, 2025-04-22:
Ich glaube der Grund ist da, das sich damit gesendete und empfange Telegramme gut unterscheiden lassen. Das ist aber nur für menschliches Mitlesen relevant. Das die bei ABL nur Holding Register haben und keine Input Register - man aber nicht jedes Register auch schreiben kann - ist für mich ein weiteres Zeichen dafür, das man dort das Rad von 0 auf neu erfunden hat und sich nicht viel mit Standards beschäftigt. Dazu geht das Ding nach wenigen Sekunden in den Standby und antwortet dann auf das erste Telegram nicht. Das ist auch mehr als nervig.