eModbus/eModbus #410: Separate Library Without TCP

EmperorArthur, 2025-05-07:
This library looks to fit my use case extremely well.  However, my application is purely Modbus RTU, and simply installing it via PlatformIO resulted in build failures.

Specifically, I have two Modbus RTU devices, both with an ID of 1, but each with a separate port/tranciever.  The bridge appears to easily allow for combining those into one virtual USB interface, while also providing additional functionality and error checking.

# Suggestion
I would suggest breaking this into two parts.  A core library, and a library that has everything but relies on external TCP libraries..

# Alternative
I ended up using a different library.

Miq1, 2025-05-07:
What is preventing you from just taking the base and RTU files to get a RTU only application?

Miq1, 2025-05-07:
A slightly more elaborate explanation is due 😉 

The essence of the bridge functionality is to transfer Modbus messages between the different protocols - mainly TCP->RTU and back. Hence a bridge normally will need both the RTU and TCP capabilities.

Your application with two devices having an ID of 1 normally would not be solved by a bridge, but by changing the ID of one of the two devices. I know of no Modbus device up to now that would not have a means to do so.

However, you are free to use another approach, as come to mind
- Use two ``ModbusClientRTU`` and one ``ModbusServerRTU`` on your ESP and do the programming to address the respective bus by hand. This way no TCP functionality will be needed at all.
- edit ``ModbusBridgeTemp.h`` to remove or comment out all references to TCP client functions. This should be sufficient to get a RTU-RTU bridge without the requirement of TCP lib functions.
- use another software as you did.

closed 2025-05-07 by Miq1