eModbus/eModbus #416: CRC check Error

Rahulkand, 2025-07-05:
// =================================================================================================
// eModbus: Copyright 2020 by Michael Harwerth, Bert Melis and the contributors to ModbusClient
//               MIT license - see license.md for details
// =================================================================================================

// Example code to show the usage of the eModbus library. 
// Please refer to root/Readme.md for a full description.

// Includes: <Arduino.h> for Serial etc.
#include <Arduino.h>

// Include the header for the ModbusClient RTU style
#include "ModbusClientRTU.h"

// Create a ModbusRTU client instance
// In my case the RS485 module had auto halfduplex, so no parameter with the DE/RE pin is required!

#define RXD2 16     // ESP32 UART2 RX
#define TXD2 17     // ESP32 UART2 TX
#define DE_RE_PIN 4 

ModbusClientRTU MB(DE_RE_PIN);

// Define an onData handler function to receive the regular responses
// Arguments are Modbus server ID, the function code requested, the message data and length of it, 
// plus a user-supplied token to identify the causing request
void handleData(ModbusMessage response, uint32_t token) 
{
  Serial.printf("Response: serverID=%d, FC=%d, Token=%08X, length=%d:\n", response.getServerID(), response.getFunctionCode(), token, response.size());
  for (auto& byte : response) {
    Serial.printf("%02X ", byte);
  }
  Serial.println("");
}

// Define an onError handler function to receive error responses
// Arguments are the error code returned and a user-supplied token to identify the causing request
void handleError(Error error, uint32_t token) 
{
  // ModbusError wraps the error code and provides a readable error message for it
  ModbusError me(error);
  Serial.printf("Error response: %02X - %s\n", (int)me, (const char *)me);
}

// Setup() - initialization happens here
void setup() {
// Init Serial monitor
  Serial.begin(9600);
  Serial.println("__ OK __");
  while (!Serial) {}
  Serial.println("__ OK __");

// Set up Serial2 connected to Modbus RTU
// (Fill in your data here!)
  RTUutils::prepareHardwareSerial(Serial2);
  Serial2.begin(9600, SERIAL_8N1, RXD2, TXD2);

// Set up ModbusRTU client.
// - provide onData handler function
  MB.onDataHandler(&handleData);
// - provide onError handler function
  MB.onErrorHandler(&handleError);
// Set message timeout to 2000ms
  MB.setTimeout(2000);
// Start ModbusRTU background task
  MB.begin(Serial2);
// The output on the Serial Monitor will be (depending on your Modbus the data will be different):
//      __ OK __
//      Response: serverID=1, FC=3, Token=00000457, length=15:
//      01 03 0C 60 61 62 63 64 65 66 67 68 69 6A 6B
//      Response: serverID=1, FC=16, Token=00000458, length=19:
//      01 10 00 21 00 06 0C 11 11 22 22 33 33 44 44 55 55 66 66
//      Response: serverID=1, FC=3, Token=00000459, length=15:
//      01 03 0C 11 11 22 22 33 33 44 44 55 55 66 66
}

// loop() - nothing done here today!
void loop() {
  
  static uint32_t Token = 1111;

  Error err = MB.addRequest(Token++, 1, READ_HOLD_REGISTER, 0x2103, 1);
  if (err!=SUCCESS) {
    ModbusError e(err);
    Serial.printf("Error creating request: %02X - %s\n", (int)e, (const char *)e);
  }

// Create request for
// (Fill in your data here!)
// - server ID = 1
// - function code = 0x16 (write multiple registers)
// - address to write = word 33ff
// - data words to write = see below
// - data bytes to write = see below
// - token to match the response with the request.
//
  static uint16_t wData[] = { 1000 };
  Serial.println(wData[0]);

  err = MB.addRequest(Token++, 1, WRITE_MULT_REGISTERS, 0x2001, 1, 2, wData);
  if (err!=SUCCESS) {
    ModbusError e(err);
    Serial.printf("Error creating request: %02X - %s\n", (int)e, (const char *)e);
  }

// Create request for
// (Fill in your data here!)
// - server ID = 1
// - function code = 0x03 (read holding register)
// - address to read = word 33
// - data words to read = 6
// - token to match the response with the request.
//
  err = MB.addRequest(Token++, 1, READ_HOLD_REGISTER, 0x2103, 1);
  if (err!=SUCCESS) {
    ModbusError e(err);
    Serial.printf("Error creating request: %02X - %s\n", (int)e, (const char *)e);
  }

  wData[0] += 10;
  delay(10000);

}

I am trying to read data from delta VFD register (0x2103H) using Modbus RTU protocol. The Master here is esp32. I am using an MAX485 Module to convert UART signals, generated by esp32, to RS485 signals and vice versa. while reading the data from VFD, i am persistently getting CRC check error OR packet length error, but very few times i get the values correctly.

Response: serverID=1, FC=3, Token=00000592, length=5:
01 03 02 07 F8 
Response: serverID=1, FC=16, Token=00000593, length=6:
01 10 20 01 00 01 
Error response: E2 - CRC check error

Error response: E2 - CRC check error
Error response: E2 - CRC check error
Response: serverID=1, FC=3, Token=000005AF, length=5:
01 03 02 08 4C

Miq1, 2025-07-05:
You may need to check your RTU wiring - CRC and packet length errors commonly are hinting to a bad wiring/termination/biasing on the RS485 side.

You may compile your code with a higher logging level, like ``-DLOG_LEVEL=6`` (or set the default ``LOG_LEVEL`` in the file ``Logging.h`` to ``LOG_LEVEL_VERBOSE``). This will give you a more detailed insight into the data going back and forth. Feel free to post the results here for us to examine.

Miq1, 2025-07-09:
Closing this as there is no reaction from TO.

closed 2025-07-09 by Miq1