eModbus/eModbus #418: setTarget function returning incorrect answer, if MT_lastTarget was not updated

RatherNoName, 2025-07-30:
Hello. I run into issue. 

1.  setTarget(192.168.1.101, 502); // Correct address
2.  setTarget(192.168.1.105, 502); // Incorrect address
3.  setTarget(192.168.1.101, 502); // Again correct address. 

The answer on the 3th call will be false, due the fact that MT_lastTarget is updated after the client is connected successfully, therefore on the 2nd call, this value is not updated.

Miq1, 2025-07-30:
I rather would like to invalidate ``MT_lastTarget`` by assigning an empty ``TargetHost`` object in the ``else`` path, where the failed connection is handled. About [here in ``ModbusClientTCP.cpp``](https://github.com/eModbus/eModbus/blob/4368b4b27f669635abba4de053fba4769dedab57/src/ModbusClientTCP.cpp#L381).

RatherNoName, 2025-07-30:
Than I guess the same would need to be added here [Here](https://github.com/eModbus/eModbus/blob/4368b4b27f669635abba4de053fba4769dedab57/src/ModbusClientTCP.cpp#L356)

Miq1, 2025-07-30:
> Than I guess the same would need to be added here [Here](https://github.com/eModbus/eModbus/blob/4368b4b27f669635abba4de053fba4769dedab57/src/ModbusClientTCP.cpp#L356)

No, that spot is handling Modbus errors happening on a functioning connection. No need to reconnect here.

RatherNoName, 2025-07-30:
Than OK for me.

Miq1, 2025-07-30:
I will try to fix that until the end of the week.

closed 2025-08-05 by Miq1