eModbus/eModbus #421: ModbusClientRTU not decoding responce in time from SLAVE for ESP32-S3

OptifySudarshanPatil, 2025-08-12:
**Describe the bug**
Im using `ModbusClientRTU`  to poll status of input channel of waveshare 8 channel DIDO module from ESP32-S3 and RS485 module .  while sending READ_DISCR_INPUT to device, im getting timeout error.

[image]

```bash
[D] 97186| RTUutils.cpp         [ 436] receive: C/[D] Received packet: @3FCF1C98/1:
  | 0000: E0                                                |.               |
[D] 97197| ModbusClientRTU.cpp  [ 283] handleConnection: Error response (1 bytes) received.
[V] Data: @3FCF1C98/1:
  | 0000: E0                                                |.               |
[D] 97207| ModbusClientRTU.cpp  [ 304] handleConnection: Response generated.
[V] Response packet: @3FCF1C98/3:
  | 0000: 03 82 E0                                          |...             |
[E] 97228| emodbus_function.h   [  69] handleError: Error response: E0 - Timeout
[V] 97645| ModbusMessage.cpp    [ 473] checkData: Check data #3
[D] 97646| ModbusClientRTU.cpp  [ 162] addRequestM: request for 03/02
[D] 97656| ModbusClientRTU.cpp  [ 245] addToQueue: RC=01
[D] 97656| ModbusClientRTU.cpp  [ 173] addRequestM: RC=00
[D] 97657| ModbusClientRTU.cpp  [ 263] handleConnection: Pulled request from queue
[D] Sent packet: @3FCF1C98/6:
  | 0000: 03 02 00 01 00 07                                 |......          |
[D] 97676| ModbusClientRTU.cpp  [ 268] handleConnection: Request sent.
[D] 99687| RTUutils.cpp         [ 436] receive: C/[D] Received packet: @3FCF1C98/1:
  | 0000: E0                                                |.               |
[D] 99698| ModbusClientRTU.cpp  [ 283] handleConnection: Error response (1 bytes) received.
[V] Data: @3FCF1C98/1:
  | 0000: E0                                                |.               |
[D] 99708| ModbusClientRTU.cpp  [ 304] handleConnection: Response generated.
[V] Response packet: @3FCF1C98/3:
  | 0000: 03 82 E0                                          |...             |
[E] 99729| emodbus_function.h   [  69] handleError: Error response: E0 - Timeout
[V] 100146| ModbusMessage.cpp    [ 473] checkData: Check data #3
[D] 100147| ModbusClientRTU.cpp  [ 162] addRequestM: request for 03/02
[D] 100157| ModbusClientRTU.cpp  [ 245] addToQueue: RC=01
[D] 100157| ModbusClientRTU.cpp  [ 173] addRequestM: RC=00
[D] 100158| ModbusClientRTU.cpp  [ 263] handleConnection: Pulled request from queue
[D] Sent packet: @3FCF1CC8/6:
  | 0000: 03 02 00 01 00 07                                 |......          |
[D] 100176| ModbusClientRTU.cpp  [ 268] handleConnection: Request sent.
[D] 102187| RTUutils.cpp         [ 436] receive: C/[D] Received packet: @3FCF1C98/1:
  | 0000: E0                                                |.               |
[D] 102198| ModbusClientRTU.cpp  [ 283] handleConnection: Error response (1 bytes) received.
[V] Data: @3FCF1C98/1:
  | 0000: E0                                                |.               |
[D] 102208| ModbusClientRTU.cpp  [ 304] handleConnection: Response generated.
[V] Response packet: @3FCF1C98/3:
  | 0000: 03 82 E0                                          |...             |
[E] 102229| emodbus_function.h   [  69] handleError: Error response: E0 - Timeout
[V] 102647| ModbusMessage.cpp    [ 473] checkData: Check data #3
[D] 102648| ModbusClientRTU.cpp  [ 162] addRequestM: request for 03/02
[D] 102658| ModbusClientRTU.cpp  [ 245] addToQueue: RC=01
[D] 102658| ModbusClientRTU.cpp  [ 173] addRequestM: RC=00
[D] 102659| ModbusClientRTU.cpp  [ 263] handleConnection: Pulled request from queue
[D] Sent packet: @3FCF1CC8/6:
  | 0000: 03 02 00 01 00 07                                 |......          |
[D] 102677| ModbusClientRTU.cpp  [ 268] handleConnection: Request sent.
```

## this is responce from RS485 sniffer connected to same bus
```bash
{03}{02}{00}{01}{00}{07}{69}{EA}{03}{02}{01}{00}{A0}{30}
{03}{02}{00}{01}{00}{07}{69}{EA}{03}{02}{01}{00}{A0}{30}
{03}{02}{00}{01}{00}{07}{69}{EA}{03}{02}{01}{00}{A0}{30}
{03}{02}{00}{01}{00}{07}{69}{EA}{03}{02}{01}{00}{A0}{30}
{03}{02}{00}{01}{00}{07}{69}{EA}{03}{02}{01}{00}{A0}{30}
{03}{02}{00}{01}{00}{07}{69}{EA}{03}{02}{01}{00}{A0}{30}
```

[image]

**Expected behavior**
it should give responce in "handleData" or "onResponse" handle

**To Reproduce**
Steps to reproduce the behavior: 

**Example code**
```c
#include "ModbusClientRTU.h"
#include "Logging.h"
#define RS485BAUDRATE 9600
ModbusClientRTU MB_RTU_CLIENT(RS485_RE_DE);

void setup(){
  RTUutils::prepareHardwareSerial(Serial2);
  Serial2.begin(RS485BAUDRATE, SERIAL_8N1, ESP_TX1, ESP_RX1, true);
  MB_RTU_CLIENT.onDataHandler(&handleData);
  MB_RTU_CLIENT.onResponseHandler(&onResponse);
  MB_RTU_CLIENT.onErrorHandler(&handleError);
  MB_RTU_CLIENT.setTimeout(2000);
  MB_RTU_CLIENT.begin(Serial2);
}

void loop()
{
  static unsigned long time_of_last_waveshare_do_blink = 0;
  if ((millis() - time_of_last_waveshare_do_blink) > 2500)
  {
    time_of_last_waveshare_do_blink = millis();
    data_ready = false;
    waveshare_do_blink_state = !waveshare_do_blink_state;
    uint16_t coil_value = waveshare_do_blink_state ? 0xFF00 : 0x0000;
    // Error err = MB_RTU_CLIENT.addRequest((uint32_t)millis(), waveshare_dido_id, WRITE_COIL, 1, coil_value);
    Error err = MB_RTU_CLIENT.addRequest((uint32_t)millis(), waveshare_dido_id, READ_DISCR_INPUT, 1, 7);
    if (err != SUCCESS)
    {
      ModbusError e(err);
      LOG_E("Error adding request: %02X - %s\n", (int)e, (const char *)e);
    }
  }
}
```

**Additional context**
Set loglevel

```
#ifndef LOG_LEVEL
#define LOG_LEVEL LOG_LEVEL_VERBOSE
#endif
```

Miq1, 2025-08-12:
Some remarks for a start:
- an ``onResponseHandler`` will invalidate any ``onErrorHandler`` or ``onDataHandler`` that may be defined. If there is an ``onResponseHandler``, others will simply not be checked any more.
- Your code snippet is missing any of the handlers, you need to write your own to have that mechanism working
- Your sniffer protocol as well as the log are showing the requests like ``{03}{02}{00}{01}{00}{07}{69}{EA}`` are in fact sent. You want to read 7 status bits, starting at address 1.
- It looks like the server is sending an answer as well: ``{03}{02}{01}{00}{A0}{30}``, meaning all 7 status bits are ``0`` at the moment.

I would suggest you reading the documentation on eModbus' ``CoilData`` type and more general the examples regarding RTU clients and their (very simple) handler functions.

Miq1, 2025-08-13:
Ok, sorry to suggest you to look at the basics first 😉 

Taking into consideration the handlers now (only ``onResponse`` will apply as I wrote) it looks like there might be a wiring issue:
- since transmitting seems to be working the RX line is worth to be scrutinized. Is the connection between the RX GPIO of the ESP32 and the adapter in place and has connectivity? Is the GPIO connected that that you used in the Serial begin call?
- are the adapter's RE and DE pins both connected to your designated toggle GPIO?

Miq1, 2025-08-17:
Looks like this is not relevant any more...

closed 2025-08-17 by Miq1