eModbus/eModbus #424: Multiple Modbus clients/threads getting the same task name

pionsys-mhs, 2025-09-26:
**Describe the bug**
We are using 2 instances of the ModbusClientRTU. Each instance uses its own pin, so no problem there. However, when the task is being created the static variable "instanceCounter" is used for the task name. This variable is incremented as soon as the ModbusClientRTU object is created, which leads to 2 tasks with the identical name of "Modbus02RTU".

**Expected behavior**
I would expect the names of the tasks to be unique to clearly differentiate them, by using a variable like "instanceIndex"  that gets the value of the static variable "instanceCounter" when the ModbusClientRTU object is being created.

Miq1, 2025-09-26:
Should not happen in fact. The base class ``ModbusClient`` constructor in ``ModbusClient.cpp`` is called implicitly when creating another client instance. That constructor will increment the ``instanceCounter`` variable - the fact that you can see the ``02`` in the task name is proving the increment was done.

Can you provide us with a task list showing the two tasks named identically?

Miq1, 2025-09-26:
Ah! Thanks for the clarification. So each instance will require a local copy of the state of ``instanceCounter`` at the time it is instantiated. Makes sense - I will see to it being fixed.